arduino-cmake/arduino-cmake #1: Can't generate sketches

JonasProgrammer, 2017-09-23:
[copy of https://github.com/JonasProgrammer/arduino-cmake/issues/7 by @Souler]

After the latest changes(e97627543505ecdc9247be595490a03b9e178028), cmake build broke (at least for me). Also, this same config works fine on version 144eccc08513026d881cb55023e0c343e7e9bc5b

**CMakeLists.txt**
```
project(myproject C CXX ASM)

print_board_list()
print_programmer_list()

link_directories(${CMAKE_CURRENT_SOURCE_DIR}/lib)

generate_arduino_firmware(myproject
	SKETCH "${CMAKE_CURRENT_SOURCE_DIR}/myproject/Sketch.ino"
	BOARD uno
	PORT /dev/ttyUSB0
	SERIAL screen @SERIAL_PORT@)
``` 

**CMake output**
```
-- The C compiler identification is GNU 4.8.2
-- The CXX compiler identification is GNU 4.8.2
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/avr-gcc
-- Arduino SDK version 1.8.4: /usr/share/arduino-sdk1.8
-- Check for working C compiler: /usr/bin/avr-gcc
-- Check for working C compiler: /usr/bin/avr-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/avr-g++
-- Check for working CXX compiler: /usr/bin/avr-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ARDUINO Boards:
--                   menu: 
--                    yun: Arduino Y
--                    uno: Arduino/Genuino Uno
--              diecimila: Arduino Duemilanove or Diecimila
--                   nano: Arduino Nano
--                   mega: Arduino/Genuino Mega or Mega 2560
--                megaADK: Arduino Mega ADK
--               leonardo: Arduino Leonardo
--            leonardoeth: Arduino Leonardo ETH
--                  micro: Arduino/Genuino Micro
--                esplora: Arduino Esplora
--                   mini: Arduino Mini
--               ethernet: Arduino Ethernet
--                    fio: Arduino Fio
--                     bt: Arduino BT
--             LilyPadUSB: LilyPad Arduino USB
--                lilypad: LilyPad Arduino
--                    pro: Arduino Pro or Pro Mini
--               atmegang: Arduino NG or older
--           robotControl: Arduino Robot Control
--             robotMotor: Arduino Robot Motor
--                  gemma: Arduino Gemma
--     circuitplay32u4cat: Adafruit Circuit Playground
--                yunmini: Arduino Y
--                chiwawa: Arduino Industrial 101
--                    one: Linino One
--                unowifi: Arduino Uno WiFi
-- 
-- ARDUINO Programmers:
--            avrisp: AVR ISP
--        avrispmkii: AVRISP mkII
--        usbtinyisp: USBtinyISP
--        arduinoisp: ArduinoISP
--     arduinoisporg: ArduinoISP.org
--            usbasp: USBasp
--          parallel: Parallel Programmer
--      arduinoasisp: Arduino as ISP
--          usbGemma: Arduino Gemma
--         buspirate: BusPirate as ISP
--            stk500: Atmel STK500 development board
--          jtag3isp: Atmel JTAGICE3 (ISP mode)
--             jtag3: Atmel JTAGICE3 (JTAG mode)
--         atmel_ice: Atmel-ICE (AVR)
-- 
-- Generating myproject
-- SKETCH_SOURCES: 
CMake Error at arduino-cmake/cmake/Platform/Arduino.cmake:1830 (generate_sketch_cpp):
  generate_sketch_cpp Function invoked with incorrect arguments for function
  named: generate_sketch_cpp
Call Stack (most recent call first):
  arduino-cmake/cmake/Platform/Arduino.cmake:505 (setup_arduino_sketch)
  CMakeLists.txt:21 (generate_arduino_firmware)

CMake Error at arduino-cmake/cmake/Platform/Arduino.cmake:2028 (message):
  Invalid source file:
  /home/barbosa/projs/myproject/firmware/build/myproject_Sketch.ino.cpp
Call Stack (most recent call first):
  arduino-cmake/cmake/Platform/Arduino.cmake:516 (find_arduino_libraries)
  CMakeLists.txt:21 (generate_arduino_firmware)

-- Configuring incomplete, errors occurred!
See also "/home/barbosa/projs/myproject/firmware/build/CMakeFiles/CMakeOutput.log".
```

JonasProgrammer, 2017-09-23:
Well, I never used the sketch functionality, but from peeking at how it's implemented, your SKETCH parameter should be the directory containing the .ino file, rather than the file itself. Using this approach, generating works for me.

@MrPointer As you refractored that stuff, could you have a look at this please? Using the directory generating works, but the include statement for Arduino.h is generated only after the setup function, not before, thus breaking the build due to undefined variable names etc.

Souler, 2017-09-23:
As someone suggested (not sure if it was you) it is actually fixed if I change the `SKECTH` from
```
	SKETCH "${CMAKE_CURRENT_SOURCE_DIR}/myproject/Sketch.ino"
```
to
```
```
	SKETCH "${CMAKE_CURRENT_SOURCE_DIR}/myproject"
```

Still a bit weird it worked with the extension in the first place.

JonasProgrammer, 2017-09-23:
Yep, it was me. As stated, this suggestion was based soley on looking at the underlying cmake script. @MrPointer reworked the sketch part there on the 1.6 feature branch, so there might have been changes there.
As it worked with the full path before, I'm still not sure whether we should take this as a bug or a breaking change (which wouldn't be too bad considering we have no official releases atm).

By the way, did you #include <Arduino.h> in your sketch? Because the official blink sketch broke for me when testing this issue due to the include being generated only after the setup function. Thus, the Makefile was generated but the build itself failed.

Souler, 2017-09-23:
No, I did not include `#include <Arduino.h>` explicitly. But I did include `#include <SoftwareSerial.h>`, which implicitly does `#include <Arduino.h>`.

Just for the sake of testing I just ran both scenarios (implicit and explicit includes) and both worked fine with the lastest commit on this repo and arduino sdk1.8

MrPointer, 2017-09-24:
@Souler Does it mean that it builds successfully now both when using the directory path and the file path?
Or did you test it just for the directory path approach?

JonasProgrammer, 2017-09-24:
At least from my testing, the file path solution is broken whilst the directory approach works. And for him apparently only the directory works as well.

The explicit/implicit stuff seems to be another bug, where sketches can't be compiled after CMake generation due to the setup() function being defined before the Arduino.h-include. With "most" sketches that include headers other than Arduino.h anyways, this shouldn't be a problem though, as their includes are properly output to the top.

Have a look at this gist which was generated from the official Blink sketch (https://www.arduino.cc/en/Tutorial/Blink): https://gist.github.com/JonasProgrammer/04e1901d8658337fec7341d1413a623b

Souler, 2017-09-24:
@MrPointer  When setting `SKETCH` to the directory where the the `.ino` file resides, it works fine. Setting `SKETCH` to the `.ino` file path itself, fails.

MrPointer, 2017-09-25:
@Souler I think I managed to fix this issue by forcing CMake to work on the directory, even when an explicit file is given. This is possible since the **generate_arduino_sketch** function finds all existing *.ino* or *.pde* files in that directory and then generates them as sketch sources.

Please have a look at f04faf8124e2bc6000a8c2a900f1d4786c5e7ef9 and tell me if it has worked for you so I would merge it further.

Souler, 2017-10-05:
It worked fine, just in case I did a diff of both outputs (both using both directory and sketch).

```
64c64
< -- SKETCH_SOURCES: /home/barbosa/projs/project/firmware/src//Sketch.ino
---
> -- SKETCH_SOURCES: /home/barbosa/projs/project/firmware/src/Sketch.ino
121c121
< [ 95%] Building CXX object CMakeFiles/project.dir/project_Sketch.ino.cpp.obj
---
> [ 95%] Building CXX object CMakeFiles/project.dir/project_src.cpp.obj

```

MrPointer, 2017-10-05:
@Souler Merging fix into master. See #9 .
Thank you for finding this bug!

closed 2017-10-05 by MrPointer