arduino-cmake/arduino-cmake #2: CMake generation fails to find Arduino SDK path

JonasProgrammer, 2017-09-23:
[copy of https://github.com/JonasProgrammer/arduino-cmake/issues/6 by @MrPointer]

## Background
A while ago I've setup my work computer (Completely offline) to work with this fork, installing **Arduino 1.8.1** complete SDK, **CMake 3.82**, **mingw-64** as my build system and **CLion** as my IDE ***On a Windows machine***. 
  
## Issue
Unfortunately, the very first **CMake** generation has failed, yelling at me that it couldn't find `Arduino's SDK path`, which was strange. At first I though it had something to do with **CLion**, so I tried to generate it simply with **cmake-gui**, but, with no success either.  
  
## Capturing the bug
After a few days of occasional debugging, I came to a conclusion that the problem is indeed with my `SDK path`, or more accurately, where I've set it.  
Due to some reasons I had to install **Arduino** on my `D:\` drive - To put in other words, my first ever attempt to install it on something other than the casual `C:\Program Files\`. I really suspect that this might be the source of the bug, with our **CMake** script not being able to handle something other than the default, casual path.  
However, I didn't had time yet to pursue the issue and fix the bug.

## Fix is on the way
No, there is no solution yet, but I'm working on it now as I have a bit more time to address it.  
Would be glad to see someone else also attempt to solve it though!

### No logs attached
Unfortunately, I don't have any logs that I can attach here, simply because it happened only at my work computer, which is, as I stated earlier, completely offline, and I can't export anything out of the work.

JonasProgrammer, 2017-09-23:
Well, I haven't tested anything with a 1.8 SDK so far, but on both of my development machines the SDK is installed to a nonstandard location on another drive, so this shouldn't be an issue.

Have you set the ARDUINO_SDK_PATH environment variable to hint the SDK path for prior to generation?

JonasProgrammer, 2017-09-23:
[comment by @MrPointer]

@JonasProgrammer Yes, I did set the variable to the appropriate path.
I'm currently working on setting up a clean VM to investigate this further.

MrPointer, 2017-09-30:
@JonasProgrammer I finally finished setting up the VM (Setting a Windows machine up-to-date is no joke...) so I could test the issue.
Well, having installed Arduino SDK to a non-standard location (D:\Arduino) as well as mingw-64 (D:\MinGW) and running a first test - CMake fails yet again with the same old issue.

Here's my modified **CMakeLists.txt** file (I use the one from the example):
```
set(ARDUINO_SDK_PATH "D:/Arduino")
set(CMAKE_TOOLCHAIN_FILE cmake/ArduinoToolchain.cmake) # Arduino Toolchain

cmake_minimum_required(VERSION 2.8)
#====================================================================#
#  Setup Project                                                     #
#====================================================================#
project(ArduinoExample C CXX)

print_board_list()
print_programmer_list()

add_subdirectory(example)   #add the example directory into build
```

I did notice something really strange, however. 
I've added a simple print line to the beginning of the toolchain file `message(STATUS "Attempt")` and found out this got printed **TWICE**. How is that even possible? Doesn't CMake processes the toolchain file only once? I absolutely have no clue of what to do next.

JonasProgrammer, 2017-09-30:
Well, as for it running twice, I noticed that as well. I have no real idea why (I'm not that deep into CMake actually), but I actually had some issues with it. More specifically I put the whole part around [this](https://github.com/arduino-cmake/arduino-cmake/blob/f3a7ebb521da945a30855307daa0b8a4b2db2b35/cmake/ArduinoToolchain.cmake#L55)  solely because some variables weren't properly restored in between runs causing all kinds of silly errors.

The same probably goes for your manually set `ARDUINO_SDK_PATH` there; apparently the value is not preserved for some run causing the path search to re-run.

That being said, I specifically added the possibility to have an ARDUINO_SDK_PATH environment (not cmake) variable, so you can set up the SDK on different machines in different places and still use the very same build file. Could you give that a try?

If that works, I'll probably just put this workaround in the documentation somewhere (in the end using environment variables is the better way to go, if you want to implement CI for your firmware, for example). I should have more spare time the next few days, I'll try to get deeper into how CMake toolchains and variables actually work and try and solve the problem afterwards; in general, this issue occurs for all arduino related variables when you try to set them manually.

JonasProgrammer, 2017-10-01:
@MrPointer Can you give  9f8d220 a try?

MrPointer, 2017-10-01:
@JonasProgrammer  I tried both of your solutions, and here are the results:

1. Defining an environment variable named `ARDUINO_SDK_PATH` doesn't solve the issue in the old codebase - The one I've been using while discussing this issue.
2. While the environment variable has still been defined I tested the changes and it worked like charm. When I noticed that the environment variable is still there, I removed it and reloaded **CMake** - And voila! It still works. If I got you right on the commit message, it's intended to work like this, storing this configuration for long-term. Is this really the case?

JonasProgrammer, 2017-10-01:
Well that's strange. The environment variable always worked for me (I implemented it because I had exactly your problem). Could you provide exact system specs maybe? I've got it working with the environment variable on both windows and the Windows Subsystem for Linux; I wasn't bothered yet to dig out my linux laptop though.

And from my understanding, it's CMake common practise to store configurations in the cache. CMake gui will actually use the cached variable from a first run to present auto-detected results and let's the user change it afterwards. If we re-detected every time, manual path overwrites would be gone as well.

If you truly need to autodiscover libraries (or the arduino SDK) again, resetting the CMake cache is the way to go.

At least that's my point of view and experience from CMake currently, feel free to discuss if you think we should change this behaviour.

MrPointer, 2017-10-01:
@JonasProgrammer I meant I have reset the cache and reloaded CMake, not just reloaded, my mistake.
That's why it's even stranger - I know about CMake storing cache and all, but it should be all gone once reset.
About that environment behavior - I'll update this comment after I'll try something.

JonasProgrammer, 2017-10-01:
@MrPointer So as far as I read your message you're also ok with the caching system that is currently implemented?

As for the environment variable: Keep in mind, that processes inherit their parent's environment variables and that system environment variables won't be refreshed during a processes lifetime (explorer.exe aside). This is especially annoying with Jetbrains Toolbox, as you have to close all instances of CLion and exit the toolbox afterwards to populate system environment variable changes.
If you're experimenting anyways, could you maybe give the old code another try with the environment variable and everything closed and reopened? I'm still puzzeled it didn't work because I didn't touch the environment variable hinting at all...

MrPointer, 2017-10-01:
@JonasProgrammer Yes, there's nothing wrong with the caching system - It works just like I expect it to.

And as for the environment variable - Your'e absolutely right about their inheritance between processes - Never thought about it, and as I do use **CLion** (Although this time without the **Toolbox**), it makes sense that resetting and reloading **CMake** didn't help.

Now about the old code behaving strangely - I think it has nothing to do with **CLion** nor **CMake** reloading.
I set the variable before I've opened **CLion**, and it hasn't work at the very 1st attempt already.

JonasProgrammer, 2017-10-01:
Ok, I'm giving up. I'll just let the old code behave strangely; if someone needs a nonstandard location they'll have to -DARDUINO_CMAKE_PATH in Cmake for it to safely work and that's it.

MrPointer, 2017-10-01:
@JonasProgrammer Agreed. No need to mess with old code after you've managed to fix it...

closed 2017-10-01 by MrPointer