arduino-cmake/arduino-cmake #4: USB VID and PID for CDC

Baycken, 2017-09-28:
I would like to implement custom VID and PID for USB CDC. I need this for the udev rules to distinguish multiple Arduino Leonardo connected to my robotics system.

The other option is to properly implemented the iSerial of the CDC, but that requires chances in the source code of arduino core which I think is more difficult.

Basically this involve adding some if statement in here.

And maybe add a PID and VID argument in generate_arduino_firmware.

I think I have an idea how to do this, just want to make sure that you think this is useful before I start to work on this.

Baycken, 2017-09-28:
@JonasProgrammer 
The VID had PID are compiler flag for the USB CDC [here](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/USBCore.cpp), so it only works for board with native USB such as Leonardo and Due. And you have explicitly define `USBCON` to make it work.

For board using FTDI chips or board using Atmega16u2/ATmage8u2, this compiler flag is simply ignored. And for those devices, the UART/USB chip properly implement iSerial, which can be used to uniquely identify the device.

JonasProgrammer, 2017-09-29:
Fine then, we should implement this then I think.
The only thing I still really want to avoid is adding generate_arduino_firmware options that only work with certain boards and silently fail on others. So either the generation process should abort when an invalid board/option combination is detected (e.g. by parsing the boards file) or we should at least have a separate generation command with advanced options that is explicitly documented for advanced use only (the current function would become the actual advanced function and it'd simply be called from a new generate_arduino_firmware with more restrictive argument extraction).

JonasProgrammer, 2017-11-17:
Will be in the upcoming v2 release as soon as #20 is merged and I can do things from there.

fraillt, 2017-11-18:
@JonasProgrammer there are two ways, to implement this functionality.
* add optional VID/PID parameters to all public API functions, OR 
* add separate function that only responsible for setting VID/PID pair for a target.
I strongly prefer the second options, because this is advanced functionality that maybe 10% percent of users cares about, and documenting this kind of feature would also be simpler :)
the usage from user perspective would look something like this:

```
generate_arduino_firmware(blink
	SRCS  blink.cpp
	HDRS  blink.h
	BOARD uno)

set_vendor_product_id(blink
	VID 0x2341
	PID 0x0043)
```

what do you think ?

fraillt, 2017-12-15:
@Baycken could you please try `develop` branch and see if it works for you?
There is a new function to customise USB properties, see `example/CMakeLists.txt`