arduino-cmake/arduino-cmake #5: 1200BPS Touch for USB CDC upload

Baycken, 2017-09-28:
Software resetting Arudino Leonardo is kinda weird. It require the serial port to be open to 1200 bps and set DTR to low, then the Arudino will reboot into bootloader mode and allow the code to be loaded to the Arduino. This step is not included in the AVRDUDE. So if I invoke -upload command for an Arduino Leonardo, I will have to manually reset the Arduino via the reset button.

This step can be automated by using a simple python script and adding a cmake custom target, but that make the port really confusing, as the Arduino in bootloader have a different VID and PID than when it is in user code mode. While they both shows up as a ttyACM device, if you use a udev rule to identify the Arduino, or if you have other ttyACM devices, then you will need 2 rules, one for communicating normally with the arduino, one for bootloader mode. And you will have to have 2 ports specified, one to kick the arduino into bootloader mode, one for uploading the code to Arduino.

The really annoying part of this is that the boards.txt only have `use_1200bps_touch` atrribute for really recent version of arduino. for older version, which board need the 1200bps kick into bootloader is hardcoded. So we cant really know if the board need to be kicked or not.

This issue and #4 are both targeted to Arduino Leonardo and I propose that we create and new `generate_firmware` function for Leonardo only to solve both of them.

JonasProgrammer, 2017-09-30:
Phew, quite some very specific issue.

While in theory we could add board-specific genration commands, I think in practise this would get messy after some boards being added to it. The PID/VID stuff from #4 being added to some advanced command made sense, but solely restricting it to a certain board would lead to either if-cascades in the actual firmware generation process or duplicated code (which would make bugfixes harder, as they have to occur for every board).

That being said, you stated you use a python script right now to do the work. Would a simple option to to specify a custom upload command fill your needs as well? You could have `UPLOAD_CMD_OVERWRITE` in `generate_arduino_firmware(_ex)` that simply gets the generated hex and eeprom files as arguments. Perhaps also with a `UPLOAD_CMD_FLAGS` argument to provide your udev rules in-place?

Baycken, 2017-10-01:
I dont think a complete command overwrite is necessary, as this is an extra step, and I am not overwriting the command for upload. A custom dependency may be the best option.

This is how I am doing that the moment:
Python:
```python
#!/usr/bin/env python

import sys
import serial
from time import sleep

arduino = serial.Serial(sys.argv[1], 1200)
arduino.setDTR(False)
arduino.close()
sleep(1)
```
 Then I added it as a custom dependency on my cmake:
```
add_custom_target(${PROJECT_NAME}_src_ros_led-touch
  COMMAND rosrun ${PROJECT_NAME} 1200bps_touch /dev/led
)

add_dependencies(${PROJECT_NAME}_src_ros_led-upload
  ${PROJECT_NAME}_src_ros_led-touch
)
```

Baycken, 2017-10-01:
For Arduino version greater than 1.0.5, this process can be automated without needing an extra flags as the `board.txt` has the `<board_name>.upload.use_1200bps_touch` set to true for all boards that uses it.

The problem with upload is that you also have to know the port to upload to, which is different that the port you communicate with.

JonasProgrammer, 2017-10-05:
This seems to be a rather 'common' issue, now that it also occurs on the Nano. But I'm still unsure on how to properly tackle this.
The reset procedure itself is clear and detecting it for newer boards shouldn't be a problem; as there'd be BOOLEAN cache variable anyways, you could still always manually set it after the generate command for the older boards.

The implementation of the reset is what bothers me. I don't really want a to add a python environment to the dependencies for this alone. `stty` and temporarily acquiring a file descriptor could do the job, I'm not sure whether there is a shell-independent way to do this though.
As I don't have access to a Nano or Leonardo right now, could you try whether you can get something running using only these utilities? I'm not even thinking of Windows for now, that'll be a pain on it's own (there seems to be something similiar with the mode command though)...

If there is a way using only standard system utility, I'm all open for putting this in. Should it really just not work without a scripting language installed, we've got to fallback to a PREUPLOAD_CMD option or something; I know it's not pretty, but for sure more pleasent than telling users they need a properly set up python environment in PATH to make the upload command work on certain boards...

Baycken, 2017-10-05:
I confirm that `stty` works for both MacOS and Linux, but with quirks.

On Mac (MacOS 10.12.6) I used `stty -f /dev/tty.usbmodem1411 1200 hub`.
On Linux (Ubuntu 16.04) I used `stty -F /dev/ttyACM0 1200 hub`

Note the different casing for `-f` and `-F`.

necauqua, 2018-06-22:
I am sorry to bring this old thing up, but you really helped me with that `stty` trigger, as I was searching exactly how to make it reset itself from the command line.
My udev rule points both product ID's to the same symlink and it works.
So for me this setup works very well:
```cmake
set(${PROJECT_NAME}_PORT /dev/arduino/micro)

generate_arduino_firmware(${PROJECT_NAME}
        SRCS ${PROJECT_NAME}.cpp
        BOARD micro
        BOARD_CPU atmega32u4
        ARDLIBS EEPROM Keyboard Mouse)

add_custom_target(${PROJECT_NAME}-reset
    COMMAND stty -F ${${PROJECT_NAME}_PORT} 1200 hup
    COMMAND sleep 1
)

add_dependencies(${PROJECT_NAME}-upload ${PROJECT_NAME}-reset)
```