arduino-cmake/arduino-cmake #10: Should the project be refactored?

MrPointer, 2017-10-05:
Hello everybody!

It's been a long time now that I feel uncomfortable with the way this project is organized. My background is mostly high-level, OOP programming, and I truly believe in *Clean Code*.
This project that we got here is far from *Clean*.
Let me explain myself further:

- The entire project consists of only 2 files: The *Toolchain* and the main *CMake script*.
- The main *script* consists of almost **3,000** lines!
- Each function has only a small set of other functions related to it, leading to extremely poor cohesion.

All these points led me to think that this project should be refactored in order to:

- Provide better support for existing features and potential bugs
- Make the project extensible so that new features would be added easily and painlessly
- Follow the `S.O.L.I.D` principles as much as functional scripting such as *CMake* can.

### TL;DR ###
Functions and macros should be extracted to separate files from the main `Arduino.CMake` script to improve code cleanliness.

What do you guys think? Should I go for it and attempt to refactor the project? Or is it just unnecessary in such an environment?

Baycken, 2017-10-05:
It will be really nice to refactor the code. I always have difficulty to find exactly how and where things get set.

It would be nice to make this not a fork of previous repos, since search does not work with forked repos on github.

JonasProgrammer, 2017-10-05:
+1 for refractoring, it takes ages to alter behaviour and Ctrl+F is the most useful navigation tool right now.

I'd just like to have @Baycken's issues fixed first (I'll try to get it next week) so we don't have to upstream-port the stuff.

MrPointer, 2017-10-06:
@JonasProgrammer I have a lot of free time now and I would like to use it to refactor the project.
I think that merging the changes shouldn't be an issue as we'll work on 2 separate branches, and the *refactor* branch wouldn't be merged to `master` until all other features gets to be refactored as well.
Yes, it will consume us some time at the stage of the merging but I think it's better than wasting time, isn't it? :)

JonasProgrammer, 2017-10-08:
@MrPointer Sorry for the delayed response, but yes, in general I think you can just start off. Perhaps keep in mind we might need two separate commands for "normal" firmware generation and "advanced" stuff that might not work across all boards. Apart from that, could you maybe push the feature branch from time to time so I could try and fix the issues on the bleeding edge, rather than doing it in a way that might not properly integrate with the refractored project?

MrPointer, 2017-10-08:
@JonasProgrammer Of course. I will constantly push the branch.

closed 2017-10-12 by MrPointer