arduino-cmake/arduino-cmake #11: CI Configuration

MrPointer, 2017-10-08:
## Continuous Integration in Arduino CMake ##
### Current Status
Currently we have CI support only for **Linux** images, specifically *Ubuntu Trusty* (14.04).
I would very much like to see support for **Windows** and **OS X** as well, since this project aims to be completely cross-platform (After all, that's what CMake is made for).

### Windows Support
I don't know much about CI, but a quick research showed me that support for **Windows** can't be achieved through our current provider, *Travis CI*. Very unfortunate, but we must find an alternative.
Searching for one, I got lost in this huge, ever-growing world, finding so many yet so few possible solutions.

I need your help on this one - What do you think is best to use? 
After reaching an agreement, I can take it further and configure a build.

### OS X Support
As **OS X** can be configured in *Travis CI* that we currently use, adding a proper configuration shouldn't be a problem at all.
I'm not that familiar with the OS to configure a build myself, so any help here would be very appreciated.

### Extended Linux Support
Although I'm quite sure that builds passing on **Ubuntu** will also pass on *e.g* **Fedora** or any rpm-based OS, we can't be sure until it will be built in such configurations.
Should we also configure builds for **Fedora**-like systems? If so, it seems impossible to do with *Travis CI*. Can you think of an appropriate alternative?

Souler, 2017-10-09:
On many of the open-source projects I follow, the **Windows** CI alternative seems to be [AppVeyor](https://ci.appveyor.com). Also, as you stated, **OSX** support can be achieved on TravisCI by tweaking the config.

About the extended Linux Support, I have no idea, never though about it, but in my opinion it's not really worth worrying about it. Our current CI linux build script makes use of the apt packages that are stated to be necessary on the `README`, the means of every user for getting said packages on its own distro are up to them, and since we don't specify any restriction on said packages versions I really think trying to support **extended Linux Support** would be more like **trying to get X package to Y distro**

MrPointer, 2017-10-09:
@Souler Thank you for your descriptive answer!
### Windows
I did checkout *AppVeyor. However, It seems to be able to run .NET code exclusively, doesn't it? I hope that I'm mistaken because it truly does seem very appropriate for the task.
### Linux
I agree with you 100%, it's just a package issue and should be out of the scope of this project.
### Summary
When you wrote that you can tackle it tonight, do you mean to tackle **Windows**, **OS X** or both?

Souler, 2017-10-10:
Travis CI with macosx support achieved [here](https://github.com/NewNetWay/arduino-cmake/blob/master/.travis.yml), take in mind that MacOSX build queue for travis.org is pretty slow.

On the windows side of the things... [here](https://github.com/NewNetWay/arduino-cmake/blob/master/appveyor.yml) is as far as I can get... I underestimated how weird things are under windows. Any help on this side is welcome.

Baycken, 2017-10-10:
What about Arduino versions? I mean, which versions of Arduino are we supporting, and should we check compilation on all supporting versions?

MrPointer, 2017-10-11:
@Souler Thanks again for supporting **OS X**! It's absolutely normal for the build to be slow since it creates a new VM on *every* new build (While the default option for e.g **Linux**, is to create a container image, which is *way* faster). Can you PR it to this repo?

As for **Windows**, simply by taking a brief look at  your [script](https://github.com/NewNetWay/arduino-cmake/blob/master/appveyor.yml), it seems to be enough, isn't it? 
It looks like you're doing exactly the same thing that we do on **Travis**, so where's the problem?

@Baycken Currently we support versions `1.6.13` and `1.8.4`. I do admit that we should also try and support versions below `1.5.0`, since this is version that changed the SDK's behavior the most.
Actually, the [Docs](https://github.com/arduino-cmake/arduino-cmake/blob/master/README.rst) says that the *minimum* supported version is **0.19**, so IMO we should test from the lower supported version. @JonasProgrammer What do you think?

MrPointer, 2017-10-27:
@Souler Reminder. Can you please refer to my comment?
@JonasProgrammer As well.

JonasProgrammer, 2017-10-28:
The question, in my oppinion, is more like whether the work to get it running on old SDKs ist worth the outcome. I have no idea, how many people are using this right now, but it's likely in the one-digit range. And we're all using somewhat recent SDK versions.
If I interpret the Arduino versioning properly, the SDK versions correspond with the IDE versions. So according to IDE 1.8.4 all older boards are still supported, thus I don't see the need to go down as low as 0.19.
Sure, there might be some breaking changes in the libraries. But honestly, anyone who's relying on that old versions has either got his own build system running (including IDE-click-the-upload-thingy-make) or at least won't feel the need for upcoming ArduinoCMake features if he cannot even be bothered to use an arduino 1.x.y SDK.
And it's not like queezythegreat's version -- which supposedly supports those older versions -- is out of the world, it's just not being updated anymore. So if you really were that one guy, who's using an ancient SDK version and looking to adapt a new build system, you should get something usable from there.

MrPointer, 2017-10-28:
@JonasProgrammer Agreed.
@Baycken Please have a look at the comment above, as it thoroughly explains out version-support methodology.

Souler, 2017-11-01:
Sorry for the delay, I have been busy lately. I've opened #15 with the OSX part.
Windows (build via AppVeyor) is giving me more trouble than expected, does someone here work with **arduino-cmake** on a windows development envirnment? If so, can you try to build any of the examples on this via powershell?

JonasProgrammer, 2017-11-02:
I'm working on a windows machine, so I could test the stuff.

JonasProgrammer, 2017-11-03:
@Souler: Have a look at this gist please: https://gist.github.com/JonasProgrammer/30eb204e54a1c4a76563b16081e2e730

As it stands, there seems to be something wrong with the AVR compiler collection delivered with 1.6.13 (haven't tried 1.8 yet). Not only does CMake fail to identify them, Windows also complained with a 'has stopped working' (thus indicating an uncaught low-level error like a segmentation fault) for both ld.exe and gcc.exe.

MrPointer, 2017-11-19:
@Souler @JonasProgrammer Any updated on this side? 
I really don't want to relay on @JonasProgrammer 's and mine ability to test thing on **Windows**. Someday we might stop supporting this project, and even though it doesn't seems very reasonable, there's a chance that there will be no one left to test things on the **Windows** side (Though it shouldn't matter that much IMO).

Souler, 2017-11-23:
I should be more free on Monday. I'll set up a PC at home with windows and try to get it working. Sorry for the silence by my side lately!

MrPointer, 2017-12-19:
Windows support through **AppVeyor** has finally been added in #36, closing.

PR #36 (2017-12-19): Add AppVeyor support (Windows CI)
As the title states.

Finally I got some time to throw into this and make it work. The `appveyor.yml` file is specially useful if we ever rewrite the `README`, since it states the bare minim needed for the proyect to run on Windows.

The proyect still needs to be enabled on the appveyor platform.

Closes #11
closed 2017-12-19 by MrPointer