arduino-cmake/arduino-cmake #16: Defining a style standard

MrPointer, 2017-11-10:
## No style standard ##
In the #13 PR an off-topic issue has been discussed, regarding the project's style standard.
As each one of us works with a different editor, whether it's a Text editor or an IDE, each also has its own style standard that it prefers over the others. As this is open source, it is totally acceptable, yet it creates some merging issues and conflicts that could easily be avoided.

## Defining one ##
The best solution to the issue would be to define a **Style Standard**, which will limit developers of the project to stick with a certain style, avoiding any unnecessary conflicts.

## EditorConfig plugin ##
There are many **Style-Enforcers**, and one of them is to use the [EditorConfig](http://editorconfig.org/) tool & plugin.
Personally, I think that it is the best way to go, even though it "forces" the developers of the project to install and plug it in their favorite editor (As far as I understand, correct me if I'm wrong). 
Many of the project in **GitHub** use it, so why shouldn't we?

I ask you guys for your opinion as well, and if you had experience with the tool - Please share your proposal for the configuration that should be used.

JonasProgrammer, 2017-11-15:
Regarding the basic formatting, I'm all for using editorconfig. It's simple and intuitive and covers the most basic flamewar potential (tabs vs. spaces, line endings and linebreaks at the file end ;) ).
It does not, however, provide a proper set of instructions regarding the programming language used. There are some proposed extensions for certain languages, but whether an editor in question implements them is another story.

Because of this, while I agree we should transform our current style (4 spaces, no tabs, LF) into an editorconfig, this won't be sufficient to satisfy other conventions like having a space between keywords and the according braces.

Personally, I'd like to have idea's codeStyleSettings.xml committed among an editorconfig. I know that at least you, @MrPointer, use it as well and I honestly don't see how a folder with a file in it would hurt non-IDEA users that much. But this is just MHO, if you don't like the proposal, we can stick to editorconfig exclusively.

MrPointer, 2017-11-17:
@JonasProgrammer Fine, you've finally convinced me lol
Can you address this issue yourself? I think that you show more understanding on the subject and also have some vision of the applied style.

JonasProgrammer, 2017-11-17:
I'll try and get it done over the course of the weekend, though I have yet to figure out how to stop CLion from adding Objective-C code styles (read: dafuq?!)  when I only changed CMake related stuff. As for the .editorconfig, keeping the current indention style, going for LF and new line at EOF should be ok, I guess?

And speaking about a 'vision' is a little too much really, exporting the current style is my main concern. I simply want to Ctrl+Alt+L without having half the file changed :D

MrPointer, 2017-11-18:
I didn't know how to replace the 'vision' word at the time, didn't mean to exaggerate lol
The Objective-C issue actually has some reason it it - After all, CLion is an IDE for multiple programming languages, with CMake probably being of the lowest 'priority'. I don't mind have some unrelated styles, because as you've already said before - It's just a small file that will probably never get updated 😆

MrPointer, 2017-11-19:
Solved in #21 .

closed 2017-11-19 by MrPointer