arduino-cmake/arduino-cmake #22: When is 2.0.0 version go live?

fraillt, 2017-11-22:
Hello, 
I want to further improve this project, by fixing multiple issues. 
Ideally i would like to work on v2, when *feature/refactoring* branch will be merged to master.
Some of the issues that i want to fix:
* executable is bigger (+10%) with this project, then with Arduino IDE (windows)
* project doesn't compile when using *pulseIn* function.
* missing multiple defines, that a lot of arduino libraries depends on
  * -DARDUINO_ARCH_(architecture) - currently only AVR, [#19](https://github.com/arduino-cmake/arduino-cmake/pull/19)
  * -DARDUINO_{build.board}
  * -DARDUINO version is incorrect (almost fixed in [#17](https://github.com/arduino-cmake/arduino-cmake/pull/17), but it seams that patch version is also should be included [firmata](https://github.com/firmata/arduino/blob/master/utility/SerialFirmata.h))
  * -DUSB_PRODUCT and -DUSB_MANUFACTURER although they are optional, it would be nice to add them as well.
* leonardo board simple doesn't compile for multiple reasons, USB support must be improved.

Question.
Should I wait for v2 and branch from master? 
I read contributing guidelines, but at this moment it still not clear to me, how should I start working :)

noseglasses, 2017-11-22:
I also initially experienced bigger builds, but on Linux. I took a look at the symbols in the binary (`avr-nm -C <firmware binary>`) and compared the output with that of the same binary build with a arduino-builder. It thus turned out that the problem is related to some missing flags that suppress, e.g. generation of exception handling. I added the following flags to gain a binary whose symbols match the arduino-builder binary.
```
-fno-threadsafe-statics
-fno-exceptions
-ffunction-sections
-fdata-sections
```
Please note that still my binaries are not identical. This may be due to a different link order of objects/precompiled static libraries. My guess is that this might be due to different padding of global variables.

If you are interested you can also take a look at a project that I am currently working on. It builds the [Kaleidoscope](https://github.com/keyboardio/Kaleidoscope) keyboard firmware using Arduino-CMake:

https://github.com/noseglasses/Kaleidoscope-CMake

MrPointer, 2017-11-22:
I believe it will be somewhere during the next week.
I'm doing my best to finish up "merging" @noseglasses 's work to the `refactoring` branch, but unfortunately I'm also releasing a big version at work, so I don't have much time to finish it this week, maybe in the weekend.
@fraillt As for your question, I think that the changes I'm integrating don't change much in the code that's currently in the `refactoring` branch, so you can start working on your issues against this branch, and compare them later just to see if something has indeed changed.

MrPointer, 2017-12-01:
Hey guys, I've just created the [v2.0.0 release branch](https://github.com/arduino-cmake/arduino-cmake/tree/release/2.0.0) as a preparation for the final version release. The [Refactoring branch](https://github.com/arduino-cmake/arduino-cmake/tree/feature/refactoring) has been completely merged to the [Develop branch](https://github.com/arduino-cmake/arduino-cmake/tree/develop) as it's finished and no longer a uniquely developed feature.  
@fraillt You can safely integrate your changes to the [Release branch](https://github.com/arduino-cmake/arduino-cmake/tree/release/2.0.0) - They seem necessary to reach a stable version.
@noseglasses Your changes seem more like an enhancement than a bug-fix, thus should be developed in a separate **feature** branch and PRed to **develop** when your'e done. They'll be integrated to the next release.

MrPointer, 2017-12-09:
Hello again everyone! @fraillt @JonasProgrammer @noseglasses @Baycken 
I think we're finally ready to release, however, we lack good documentation.
I absolutely hate the current form of documentation, listing everything in the **Readme** file - It's wrong on so many levels, but it's the only thing we've got so far. I think that we should create a new documentation based on *Wiki Pages*, copying most of the existing docs there, but it surely will take some time.
The thing is it can take too much time and create an unnecessary delay to the release of *v2.0.0*. What do you think? Should we delay the release until a proper documentation is created or release now and take care of the documentation later?

fraillt, 2017-12-10:
I would say, that currently most important is to focus on releasing new version first, and solving remaining issues. 
I do agree that documentation could be better, but we could focus on it when we no more major issues.

JonasProgrammer, 2017-12-10:
Tbh, I'd go with @fraillt here.

Everybody likes documentation but nobody likes documenting. And given that people were able to use this in the past, even with the little documentation available, I think this shouldn't be too much of an issue here.

With 2.0 going live, we can see how it does in the field. Documenting everything can be done in parallel. And if people open up issues that are not bugs but rather misuse due to lack of or inaccurate documentation, we can turn this into a documentation issue.

MrPointer, 2017-12-11:
@fraillt @JonasProgrammer Thank God you both think this way, cause I do too :)
If so, I make some final refinements and we'll be releasing right now!

MrPointer, 2017-12-13:
See #33.

closed 2017-12-13 by MrPointer