arduino-cmake/arduino-cmake #27: Feedback wanted: possible integration with a library manager?

kigster, 2017-12-04:
Hey guys, I am working on the project called "Arli" — it's a simple command line tool for managing Arduino library dependencies with each project.

It works much like Bundler does, if you are familiar with the ruby world, and it is written in Ruby also. But you don't need to know any ruby to use it.

You can search for libraries using the published Arduino database, and install either individual libraries (by name, url, etc) or you can run `arli bundle` which reads `Arlifile` — a file that documents library dependencies. 

The tricky part is that libraries must be placed in the exactly named folders, but the ZIP urls in the database have different names, and so you almost always must manually rename library folder to something matching the header file.  Well, **Arli does that automatically**. So that's a pretty solid value-add, in my opinion.

During the installation, Arli writes out the ".lock" file, and it supports multiple formats. I intended for this to include `cmake` format, so that Arli can be used as a library manager for the `arduino-cmake` project. My CMake skills are still at the beginner, but I was hoping to get some help and maybe you guys can help me figure out exactly what goes into the `Arlifile.lock.cmake` which can be included in the main `CMakeLists.txt` file.

This is my current attempt:

First, this is the "source" `Arlifile` — as you can see most of the libraries are specified by name, but one is through a github URL:

```yaml
---
dependencies:
- name: "Adafruit GFX Library"
- name: "DS1307RTC"
- name: "Adafruit LED Backpack Library"
- name: "Adafruit Unified Sensor"
- name: "DHT sensor library"
- name: "OneButton"
- name: "SimpleTimer"
  url: https://github.com/jfturcot/SimpleTimer.git
- name: "Time"
```

After Arli bundles these libraries to the custom libraries folder, it writes out this file: `Arlifile.lock.cmake`:

```cmake
# vi:syntax=cmake
set(ARLI_LIBRARIES)
# Library Adafruit GFX Library, version 1.2.2, url: http://downloads.arduino.cc/libraries/github.com/adafruit/Adafruit_GFX_Library-1.2.2.zip
prepend(ARDUINO_ARLI_LIBS ${ARDUINO_ARLI_LIBS} Adafruit_GFX)
# Library DS1307RTC, version 1.4.0, url: http://downloads.arduino.cc/libraries/github.com/PaulStoffregen/DS1307RTC-1.4.0.zip
prepend(ARDUINO_ARLI_LIBS ${ARDUINO_ARLI_LIBS} DS1307RTC)
# Library Adafruit LED Backpack Library, version 1.1.6, url: http://downloads.arduino.cc/libraries/github.com/adafruit/Adafruit_LED_Backpack_Library-1.1.6.zip
prepend(ARDUINO_ARLI_LIBS ${ARDUINO_ARLI_LIBS} Adafruit_LEDBackpack)
# Library Adafruit Unified Sensor, version 1.0.2, url: http://downloads.arduino.cc/libraries/github.com/adafruit/Adafruit_Unified_Sensor-1.0.2.zip
prepend(ARDUINO_ARLI_LIBS ${ARDUINO_ARLI_LIBS} Adafruit_Sensor)
# Library DHT sensor library, version 1.3.0, url: http://downloads.arduino.cc/libraries/github.com/adafruit/DHT_sensor_library-1.3.0.zip
prepend(ARDUINO_ARLI_LIBS ${ARDUINO_ARLI_LIBS} DHT)
# Library OneButton, version 1.2.0, url: http://downloads.arduino.cc/libraries/github.com/mathertel/OneButton-1.2.0.zip
prepend(ARDUINO_ARLI_LIBS ${ARDUINO_ARLI_LIBS} OneButton)
# Library SimpleTimer, version , url: https://github.com/jfturcot/SimpleTimer.git
prepend(ARDUINO_ARLI_LIBS ${ARDUINO_ARLI_LIBS} SimpleTimer)
# Library Time, version 1.5.0, url: http://downloads.arduino.cc/libraries/github.com/PaulStoffregen/Time-1.5.0.zip
prepend(ARDUINO_ARLI_LIBS ${ARDUINO_ARLI_LIBS} Time)
set(ARLI_LIBRARIES $ARLI_LIBRARIES PARENT_SCOPE)
```

Anyway, none of this is urgent, but if you feel like this collaboration can be fruitful, I would be very excited to work on this together.

Konstantin

MrPointer, 2017-12-04:
Hi @kigster , I looked at your project and read what you've written here and it all seems very interesting, but unfortunately I don't have enough time right now to collaborate or give any feedback.
I do hope that someone else here will have what I don't, and of course wish you the best of luck developing it further! And besides, if there's any feature your project requires from us, feel free to open a new issue and discuss about it, we'll be happy to help!

kigster, 2017-12-04:
@MrPointer — thanks so much for taking the time to review and reply. I know time is one of. E most precious and yet limited resource we all have :)

I’m pretty sure I can finish this myself, and perhaps I can use this issue to post short questions, etc. 

My plan is the following:
 * Arli will generate a CMake include file that will build each library as a static .a library, using ArduinoCmake, and then the libs must be linked with during the last step to build_firmware.
 * Arli will have a be command that generates a brand new Arduino project, with CMake tool chain, and `Arlifile` and possibly `Arli.cmake` library as well that would bridge the generated lock file and Arduino-cmake toolchain.
 * At some point Arli could even be used to build and compile the project, as well as to upload it, as a wrapper for `mkdir build; cmake ..`. 

But at the moment I’m going to focus on the best way to generate an include file that builds each library as a static lib, using the local to the project folder `./libraries`.

kigster, 2017-12-05:
@MrPointer — thanks so much for taking the time to review and reply. I know time is one of the most precious and yet limited resources we all have :)

I’m pretty sure I can finish Arli - CMake integration myself, but perhaps I can use this issue to post short questions, should I get stuck? Another option is fo me to invite you guys to the [Gitter Chat room I created for this](https://gitter.im/arduino-cmake-arli/Lobby). Let me know if this would be OK with you.

My current plan is the following:

 * Arli will generate a CMake include file called `Arlifile.cmake` that will add all library folders, and then build each library as a static .a file using macros in `arduino-cmake`, and so then user must simply add a generated variable to the linking step of the `build_firmware`.

 * Arli may eventually offer a command to auto-generate a brand new Arduino project, initialized with both `Arlifile, and the `arduino-cmake` toolchain, and perhaps the installer/validator bash script.

 * At some point Arli could even be used to build and compile the project, as well as to upload the firmware. Basically it would wrap the `arduino-cmake` project in a limited CLI API.

Such is my "dream", and I am gonna try one way or another to get to at the very least clean integration with the `arduino-cmake` project, and welcome any help, but also OK without it. One way or another, I really appreciate having `arduino-cmake` exist allowing people like me to use my preferred tools of choice, such as CLion IDE or Atom, etc. Many thanks!

MrPointer, 2017-12-21:
@kigster Finally had some time to review your work more thoroughly.
I'm having a bit of a hard time understanding where is your problem, and what keeps you from using the `cmake` **lock-file**.
If I understand correctly, you simply need to include the lock file in a desired `CMakeLists` file and reference the libraries pointed out by that file. I've created a [gist](https://gist.github.com/MrPointer/95cc3c09a06a348efce15feab5e51f25) that might help you. It contains 2 files:
1. **Renewed lock file example** - No need to `PARENT_SCOPE` libraries variable at the end as the file would simply be included/inlined. In fact, the `ARDUINO_LIBRARIES` variable isn't needed at all.
2. An example of a possible `CMakeLists.txt` which includes the lock file. Note that you must either pass the full path to the `Arli lock file` or add the file's directory to CMake's modules lists for the `include` command to work. After that, you simply reference the `ARDUINO_ARLI_LIBS` as the list of libraries to include in the firmware, as it has been defined earlier by the `include` command.

I hope that I understood you correctly, but if not, feel free to discuss about it.
BTW, you wouldn't mind to transfer the issue to your repository, would you? I feel like this one doesn't has anything to do with `arduino-cmake`, and seems to be somewhat off-topic.

kigster, 2017-12-21:
Is there a way to transfer the issue automatically? Or do you mean, copy/paste comments? 

Either way — thank you for your help and I will move the discussion over.

MrPointer, 2017-12-21:
Not one that I'm aware of, at least.
Just put a link there, probably will be better than just copy-pasting.

kigster, 2017-12-23:
Closing issue, as the discussion is moving to [arli issue](https://github.com/kigster/arli/issues/3) or alternatively this [chat board](https://gitter.im/arduino-cmake-arli/).

closed 2017-12-23 by kigster