arduino-cmake/arduino-cmake #32: Can't use Servo library

Nihlus, 2017-12-09:
Using the latest version of arduino-cmake from the `master` branch, I'm unable to use the `Servo` library that comes with the Arduino SDK. This is due to the lack of a defined processor architecture (in the case of my code which runs on an Uno, ARDUINO_ARCH_AVR).

If I manually define this in my CMake file, the project compiles but will not link (undefined reference to etc). If I manually add the correct servo source file as a part of my firmware in the CMake file, the project compiles as expected.

---
System information
Arduino SDK 1.8.0
arduino-cmake (master branch, `79dc24f `)
Linux
CLion 2017.3, Build #CL-173.3727.114
[compilation.log](https://github.com/arduino-cmake/arduino-cmake/files/1544780/compilation.log)

fraillt, 2017-12-09:
This issue is known, and i was trying to make it to 2.0.0 but unfortunately it was too late.
At beginning  of next week i'll try to make a pull request for future release, and hopefully it will soon be available on develop branch.
Stay tuned :)

Nihlus, 2017-12-09:
Superb, thank you :)

MrPointer, 2017-12-09:
@Nihlus Thank you for contributing the issue, it means a lot to us!
@fraillt Didn't we fix it already? Take a look at #23. It allows users to declare custom platforms, including architectures by setting the `PLATFORM_PATH` or `PLATFORM_ARCHITECTURE` variables.
What seems even stranger to me is that @Nihlus uses the default **avr** architecture, and it still doesn't work. I must be missing something...

fraillt, 2017-12-10:
`PLATFROM_PATH` ant `PLATFORM_ARCHITECTURE` is to be able to use non-standard avr core libraries.
This library doesn't compile for other reasons:
* `ARDUINO_ARCH_<architecture>` is not defined
* `-fno-threadsafe-statics` is not added, which in turn requires C++11 standard.

fraillt, 2017-12-11:
@MrPointer @JonasProgrammer could you merge `release/2.0.0` to `develop`? I want to make a pull request to develop, but would like to avoid merge conflicts :)

MrPointer, 2017-12-11:
@fraillt I could, but this is just not how things are done in a **GitFlow** environment. The `release` branch would be automatically merged to the `develop` branch once it's actually releases (Meaning it's also merged to `master`). Playing with those merges now will not play nicely when reviewing Git's history at a later stage, and will also break the flow.
I suggest you to merge anyway, and let us deal with conflicts later, and together if necessary.

About your previous comment, I didn't know that the `ARDUINO_ARCH_<arch>` flag exists. I truly am no expert in Arduino, I just happen to manage this project for now. Should it be resolved using the `PLATFORM_ARCHITECTURE` variable that I have mentioned?

fraillt, 2017-12-12:
Since we're finally releasing 2.0.0, so I'll just wait when develop branch will be updated :)
My implementation depends on the changes in `release/2.0.0` so I already merged it to my branch :( 
... so I'm afraid that if I do pull request to current `develop` branch it will be hard to track what my actual changes are.

Regarding `ARDUINO_ARCH_<arch>` flag, yes, I implemented it using `PLATFORM_ARCHITECTURE`, in  `RegisterSpecificHardwarePlatform.cmake` function, by caching architecture for each vendor/core.

MrPointer, 2017-12-14:
@fraillt `develop` has been updated to the latest release now - Fell free to PR.

fraillt, 2017-12-15:
@Nihlus could you please try `develop` branch, and see if it works for you?

Nihlus, 2017-12-15:
@fraillt I'm afraid not - the same issue is present. In addtionl, the latest commit from the `develop` branch causes another issue, I'm afraid, which precludes me from using it at all. The `link_directories` appears to have stopped working, and my includes are no longer resolving properly.

fraillt, 2017-12-15:
I tried [your project](https://github.com/Nihlus/Skalleper) and it works for me :)
I copied `setup` and `loop` from `Skalleper.ino` to `Skalleper.h`, commented out usages of your Chrono and StandardCplusplus libraries, and it worked, everything compiled including servo library.
I used the following `CMakeLists.txt` configuration.

```
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(PROJECT_NAME Skalleper)
project(${PROJECT_NAME} CXX C ASM)

set(CMAKE_CXX_STANDARD 11)
set(ARDUINO_DEFAULT_BOARD uno)
set(ARDUINO_DEFAULT_PORT /dev/ttyACM0)

generate_arduino_firmware(Skalleper
        SRCS Eye.cpp Skalleper.cpp
        HDRS Eye.h Skalleper.h
        BOARD uno)

set_target_properties(Skalleper PROPERTIES LINK_DIRECTORIES "/home/jarl/Arduino/libraries/Chrono;/home/jarl/Arduino/libraries/StandardCplusplus")
```

but I notices what should be changed, I should avoid using target_xxx_xxx CMake functions, because it can only be used once, and most likely users will want to use them :)

Nihlus, 2017-12-15:
I did the opposite - I commented out all references to the external sources, and only used the sketch file (in a new firmware target that only included the sketch). The servo library would not link in that instance.

fraillt, 2017-12-15:
I think, that when you use SKETCH you can't use your other sources, so you'll get errors at link time for `Skalleper` class. Maybe I'm wrong, and someone can correct me on this. 
Have you tried my way, did it worked ? :) 
Btw, from what project are these `Chrono` and `StandardCplusplus` includes, is it somewhere on the github ?

aog2000a, 2018-01-03:
At least using the current head of the "develop" branch, i needed to add
`set(Servo_RECURSE True)`
besides the already mentioned define:
`add_definitions(-DARDUINO_ARCH_AVR)`
for the build to succeed.

MrPointer, 2018-01-05:
@aog2000a This has been known actually and listed in the **README** (Our current source of documentation as well). However, I did notice that this issue comes up very often lately, so I've decided to list **ALL** recursive "standard libraries" instead of just a portion of them, so that libraries like `Servo` are automatically defined to be `RECURSE`.
This change is already available in the `develop` branch as of now :)

aog2000a, 2018-01-05:
@MrPointer Thanks! yes, i took the info from the readme. My comment here was more "for completeness" regarding the title of this post: "Can't use Servo library", in case other people find this via "googling".

MrPointer, 2018-01-06:
@aog2000a Got it, thanks!

MrPointer, 2018-01-29:
@Nihlus Whats up? Just a reminder that you owe us an update on your status 😉

Nihlus, 2018-01-30:
@MrPointer I'm still unable to test the library loading fixes, due to the other blocking issue where adding new link directories aren't recognized.

As for the libraries, they're available here:
https://github.com/SofaPirate/Chrono
https://github.com/maniacbug/StandardCplusplus

MrPointer, 2018-01-31:
@Nihlus Well, the **Chrono** and **StandardCplusplus** libraries are not recursive at all, so I can see why setting the `RECURSE` flag on them didn't work.
Could you post your `CMakeLists.txt` file's content please?

Nihlus, 2018-01-31:
Certainly.

[CMakeLists.txt](https://github.com/arduino-cmake/arduino-cmake/files/1683367/CMakeLists.txt)

MrPointer, 2018-02-02:
I got a look at it and there are few things that caught my eye as being the potential problem:

1. Why do you use `include_directories` and then `link_directories`? 
In our case only `link_directories` should be used.
2. All the `CXX` flags your'e setting are already set by default by **Arudino-CMake**.
3. Whats is the purpose of the `add_definitions(-DARDUINO_ARCH_AVR)` line?
There's a comment saying it's a workaround for incorrect arch, but I can't seem to understand why such a problem would exist.
4. The `SRCS` parameter in the `generate_arduino_firmware` function is set to an `.ino` file, which isn't considered as a valid C/C++ file.
Moreover, you attempt to use it also in the sketch, which creates a kind of a duplicate that I'm not sure is valid.

Nihlus, 2018-02-02:
1. I use CLion, which relies on it for static inspections and autocomplete.
2. They weren't in the past :) Glad to hear they're there by default now!
3. The incorrect arch workaround is for the Servo library. If it is not defined manually by me, it fails on build with the following error:

```
In file included from /home/jarl/Programming/Skalleper/Skalleper.h:26:0,
                 from /home/jarl/Programming/Skalleper/Skalleper.cpp:24:
/opt/arduino-1.8.1/libraries/Servo/src/Servo.h:69:2: error: #error "This library only supports boards with an AVR, SAM or SAMD processor."
 #error "This library only supports boards with an AVR, SAM or SAMD processor."
  ^
```

4. Without including the sketch as both a sketch and a source file, CLion once again does not recognize it as a source file included in the project. If it is not set as the sketch, the build fails with the following error:

```
libuno_CORE.a(main.cpp.obj): In function `main':
/opt/arduino-1.8.1/hardware/arduino/avr/cores/arduino/main.cpp:43: undefined reference to `setup'
/opt/arduino-1.8.1/hardware/arduino/avr/cores/arduino/main.cpp:46: undefined reference to `loop'
```

MrPointer, 2018-02-02:
I also happen to use CLion but I never heard of it. It should provide those features on successful linkage.

Have you updated **arduino-cmake** to it's latest version? I'd try to clone the `develop` branch as it has introduced many important improvements and bug fixed since the latest stable release, 2.0.2.

Nihlus, 2018-02-02:
Yes, I'm currently using the latest `develop` branch.

MrPointer, 2018-02-03:
@noseglasses Any idea?

noseglasses, 2018-02-04:
Sorry, I suppose I can't help much. 
The only idea I had is that CLion might be fooled by putting stuff in a function that is never called. This is based on the assumption that CLion does its own parse of your CMake files. If the parser is simple enough, it does not check if a function is actually called. 
This might help you with issues 1. and 4. But I am not a CLion user, so this is just a wild guess.

MrPointer, 2018-02-06:
@Nihlus This is going to sound a bit foolish, but have you reset CMake's cache and reloaded it after pulling the `develop` branch?

Nihlus, 2018-02-06:
Yes, I have.

Nihlus, 2018-05-17:
As with many things, it turns out the remaining problems were due to a PEBKAC error. I was specifying the full path to the libraries in question in `link_directories`, not the containing folder. All is well!

closed 2018-05-17 by Nihlus