arduino-cmake/arduino-cmake #34: Support for additional boards

robotdad, 2017-12-13:
Just stumbled across this project, it looks great. I just tried the 2.0 branch but it seems neither it or the current branch support additional boards installed via adding additional board manager urls, for example  ESP8266 support. Is this something that is already on your radar?

robotdad, 2017-12-13:
Was just playing around a bit more and realized only boards installed in the Arduino IDE directory are detected, at least on Windows. Boards installed later, whether through additional board urls or official ones, are installed on Windows in %localappdata%/Arduino15/packages.

MrPointer, 2017-12-13:
Thanks for contributing the issue, we really appreciate it!
Personally I didn't know about it, but it's definitely something that we'll consider to implement in the next version. I see a lot of benefit in doing so, including to my colleagues who work with various ESPs as well.
I'll keep this issue open for and return to it once we'll be ready to address. I hope that you'll still be around to provide us some more information about it :)

robotdad, 2017-12-13:
I may take a crack at it myself ver the holidays if all that is needed is an ifdef for Windows and specifying an additional directory.

MrPointer, 2017-12-13:
LGTM, though I'd still like to know more on the subject.

fraillt, 2017-12-14:
I think that currently only AVR boards are supported.
While in theory, you can rewrite `ArduinoToolchain.cmake` to add support your board, but there will be some issues:
* compile/link flags are hardcoded only for AVR (this project doesn't parse platform.txt)
* boards parameters are not customized enough, only CPU can be customized.
* auto-library detection doesn't care what board architecture is used (e.g. servo library has different implementations for AVR and ESP8266)

There are some of the issues that came to my mind, it could be more.

MrPointer, 2017-12-16:
@fraillt If so, what does `RegisterCustomHardwarePlatform` is used for?
Again, I'm not an expertise on the subject - Does a different board type (Such as ESP) necessarily means that it has its' own platform? If not, how does it differentiate then from "typical" boards such as the **uno, nano, etc**?

fraillt, 2017-12-18:
`RegisterCustomHardwarePlatform` no longer exists, it was refactored to script `RegisterSpecificHardwarePLatform`, however, for ESP it will not work.
ESP has different tools, different link libraries, different compile flags, etc... and all of these are hardcoded in multiple places in our project. 
It would be a big work to support all of it, and I think we would need to create an app/script that would generate CMakeLists.txt file with all the configuration parsed from `platform.txt`. 
Probably Java application would make sense, because Arduino requires it anyway, Python could be an alternative, but it is not by default on all OS.
On the same note, I was thinking maybe we can create a list of known issues in one place, and discuss priority and implementation details?

MrPointer, 2017-12-21:
@fraillt Of course, I meant the script `RegisterSpecificHardwarePlatform` as I'm the one who converted it in the first place 😆 
As for your proposition, I'm against any involvement of other tools/languages/dependencies other than **CMake** in this project, or in sub-projects. I don't believe that parsing the `platform.txt` file would be such a hard task that couldn't be accomplished with CMake alone.
And yes, we should do that other thing you proposed - Discussing issues and prioritizing. I will attend to this soon.

robotdad, 2017-12-21:
I for one would lose interest in this if it took dependencies on anything beyond cmake. This discussion of what the issues are is very informative, thanks. I have not had a chance to start digging into this yet myself.

BTW, on the Mac additional boards go to $HOME/Library/Arduino15/

noseglasses, 2017-12-22:
I agree. We should definitely try to do as much as we can based on CMake. Using external tools alway causes problems with finding the appropriate paths and executables on all supported platforms. That can be a nightmare.

fraillt, 2017-12-29:
Using external tools is not necessary a bad thing, it could be hidden from the user anyway.
Arduino IDE has a bundled java-runtime, so if we build something with java, users don't even have to know about it, as long as they provide correct path to Arduino (which they must do it anyway). Furthermore, such a tool could be used by Arduino itself.
Image, you open Arduino IDE, select your board, click on `tools->generate cmake project` and voila!
Wouldn't that be great? :)

MrPointer, 2017-12-29:
@fraillt What if the Ardunio SDK could be downloaded without the IDE? then you would run into a serious problem, and we don't want that...
Like @robotdad has said - Many people will loose interest in the project because of this. This isn't something that difficult to achieve with CMake, we just need to focus on it when the basics will become stable.

And yes, integrating this project in the Arudino IDE would be great but it's absolutely their responsibility and decision - We shouldn't depend on it, ever.