arduino-cmake/arduino-cmake #45: Add an example that uses a third party library

kigster, 2017-12-23:
The current example is fully self-contained, while most real Arduino projects rely on libraries.

Can a new example be added, that relies on some third party library? 

It would be particularly useful (to me at least) if the example had a configurable location of the libraries folder, that can be set to a global namespace (ie. `~/Documents/Arduino/Libraries`) or it could be a `libs` or `libraries` folder inside the project itself.

It wasn't clear to me from README how to build a sketch that relies on a third-party library, as well as (perhaps) a built-in library like Wire.

MrPointer, 2017-12-23:
You're absolutely right. In fact, we're currently undergoing some re-documentation processes, striving at better documentation using **GitHub Pages**, rewriting the **README** to contain only useful information, navigating to various pages when required.

Let's get to the point.
The way **Arduino-CMake** links external libraries today is actually written in the **README** file, but probably not as good as the original authors thought it was.
Quoting the documentation, using your requested library `Adafruit NeoPixel`:

> An Arduino library is any directory which contains a header named after the directory, simple. Any source files contained within that directory are part of the library. Here is a example of library a called Adafruit NeoPixel:

```
Adafruit_NeoPixel/
  |-- Adafruit_NeoPixel.h
  |-- Adafruit_NeoPixel.cpp
  `-- OtherLibSource.cpp
```
Now the documentation states the following:

> To incorporate a library into your firmware, you can do one of three things:
1.Place the library next to the default Arduino libraries (located at ${ARDUINO_SDK}/libraries)
2.Place the library next to the firmware configuration file (same directory as the CMakeLists.txt)
3.Place the library in a separate folder and tell Arduino CMake the path to that directory.

However, this documentation is **incorrect**, at least partially.
To use the library you should just set the `target_ARDLIBS` variable to the name of your desired library, in this case `Adafruit NeoPixel`, where `target` is usually the name of your project (The one you've set in the `project` function). 
That's it. **Arduino-CMake** will take care of the rest and will find the library automatically.
Do note that in order to include its' header in your code, you should list the full path to the library's header file. For example: `#include "libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.h"`
Also note that in order for this to work, you should place the `libraries` directory in the same location as your `CMakeLists.txt` file, or in the SDK's libraries directory, otherwise it wouldn't work.

I hope I've been clear enough. Good luck! :)

kigster, 2017-12-23:
Gotcha. Thanks for all the help.

I have my first integration project working now, please check out [arli-cmake](https://github.com/kigster/arli-cmake) which adds a couple of cmake helpers to avoid doing the long includes.

kigster, 2017-12-24:
I have completed my project integration, and now you can do the following:

```bash
$ gem install arli
$ arli generate MyProject --workspace ~/arduino/sketches
$ cd ~/arduino/sketches/MyProject
$ bin/setup
$ cd src 
$ rm -rf build && mkdir build && cd buid
$ BOARD_NAME=nano BOARD_CPU=atmega328p cmake ..
$ make; make upload
```
It will automatically detect the USB serial port (as long as there is only one), and use Arli to download and install all of the dependent libraries.

This relies on three projects:
 * [`arduino-library`](https://github.com/kigster/arduino-library) is a Ruby gem that handles searching and downloading JSON database
 • [`arli`](https://github.com/kigster/arli) is the command line Ruby gem that provides most of the functionality here; and
 • [`arli-cmake`](https://github.com/kigster/arli-cmake) — a project that bridges `arduino-cmake` and `arli`, by providing some additional CMake helpers, and integration with `Arlifile.cmake`.

kigster, 2017-12-25:
Please checkout this project — [WeatherPod](https://github.com/kigster/weather-pod) which was built using `Arli` project generator, that uses `arli-cmake` repo, which in turn uses this repo :)

MrPointer, 2017-12-25:
Looks great! I'm very happy to see it finally works!
Glad I could help a bit :)

closed 2017-12-25 by MrPointer