arduino-cmake/arduino-cmake #49: Support for referencing other variants in boards.txt

aog2000a, 2018-01-02:
The **boards.txt** file allows the following syntax
`pinoccio.build.core=arduino:arduino`
to reference a core e.g. from the main Arduino SDK (see [https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#referencing-another-core-variant-or-tool](url) for the detailed description). Currently arduino-cmake chokes on that syntax, since it preserves the full variable and can't find the corresponding folder.

My proposal would be to start removing the "VENDOR_ID:" part of the value, since multiple SDK locations are currently not supported (or am i worng?). That would allow locating the core by setting the <BOARD>_CORES_PATH variable.

MrPointer, 2018-01-05:
Sorry for being so late, have been very busy these days.
Actually this issue has been solved even before releasing **version 2.0.0**. If I recall correctly and understand your meaning well, #23 states exactly what you're trying to accomplish. 

Unfortunately our documentation is still a mess that should be corrected ASAP, and will be once I'll have more time to deal with it. For now, please relate to the [Register Custom Hardware Platforms](https://github.com/arduino-cmake/arduino-cmake/blob/develop/README.rst#advanced-options) section located just above the link (As I said, messy docs...) in the **ReadMe** file for further info.

Let me know if it's what you were looking for.

aog2000a, 2018-01-05:
@MrPointer well, yes i'm aware of the ReadMe file and contents, but i could not manage to cmake/build running by just "registering a custom platform". I needed not only the patch i posted above, but also several other "unexpected" variable definitions not mentioned in the docs. I've come further but it is still failing at linking, i guess i still need to find the right `set(xxxx_RECURSE True)` variables, but perhaps there is more.
Will post details later.

aog2000a, 2018-01-06:
@MrPointer the details of the problem, as promised.

Concretely i talk about getting code for the "Pinoccio" platform to run (based on ATmega256RFR2, see https://pinocc.io/solo for  details). The platform files are: http://pinocc.io/bootstrap/pinoccio-firmware.zip 

For instance the "LEDBlink" example inside will work without issues in the Arduino IDE.

In arduino-cmake i can't get it to work "out of the box".

Steps to reproduce:

- work in /tmp/arduino:
```
    mkdir /tmp/arduino
    cd /tmp/arduino
``` 

- extract the Arduino SDK (1.8.5)
``` 
tar xfJ /tmp/arduino-1.8.5-linux64.tar.xz
``` 

- clone the arduino-cmake repo using the *develop* branch:
``` 
    git clone --branch develop https://github.com/arduino-cmake/arduino-cmake.git
``` 

- extract the Pinoccio platform files
``` 
    unzip /tmp/pinoccio-firmware.zip
``` 

- run
``` 
    export ARDUINO_SDK_PATH=/tmp/arduino/arduino-1.8.5
    export ARDUINO_CMAKE=/tmp/arduino/arduino-cmake
    export PINOCCIO_PATH=/tmp/arduino/pinoccio
``` 

- copy the LEDBlink pinoccio example, just to have it more at hand
``` 
    cp -r pinoccio/libraries/pinoccio/examples/LEDBlink blink
``` 

- switch to the blink folder and create a CMakeLists.txt file

> 
> set(PLATFORM_PATH          $ENV{PINOCCIO_PATH}/hardware)
> set(PLATFORM_ARCHITECTURE avr)
> 
> set(CMAKE_TOOLCHAIN_FILE $ENV{ARDUINO_CMAKE}/cmake/ArduinoToolchain.cmake)
> 
> cmake_minimum_required(VERSION 3.0)
> project(ArduinoExample C CXX ASM)
> 
> print_board_list()
> print_programmer_list()
> 
> set(ARDUINO_DEFAULT_BOARD pinoccio)
> set(ARDUINO_DEFAULT_BOARD_CPU atmega256rfr2)
> set(ARDUINO_DEFAULT_PORT /dev/pinoccio)
> 
> generate_arduino_firmware(blink
> 		SKETCH LEDBlink.ino
> 		NO_AUTOLIBS)
> 

- run
``` 
    mkdir build
    cd build
    cmake ..
``` 

With the above CMake file i get immediately:

>
> -- Arduino SDK version 1.8.5-2017.01.09: /tmp/arduino/arduino-1.8.5
> CMake Error at /tmp/arduino/arduino-cmake/cmake/Platform/Core/VariableValidator.cmake:17 (message):
>   ARDUINO_CORES_PATH not set: Invalid Arduino SDK path
>   (/tmp/arduino/arduino-1.8.5).
>

- so, i define the requested variable, just to see that it wants ARDUINO_BOOTLOADERS_PATH. It goes on reqesting several variables, until I end up with the following CMakeLists.txt file:

>
> set(ARDUINO_CORES_PATH       $ENV{ARDUINO_SDK_PATH}/hardware/arduino/avr/cores)
> set(ARDUINO_BOOTLOADERS_PATH $ENV{ARDUINO_SDK_PATH}/hardware/arduino/avr/bootloaders)
> set(ARDUINO_BOARDS_PATH      $ENV{ARDUINO_SDK_PATH}/hardware/arduino/avr/boards.txt)
> set(ARDUINO_PROGRAMMERS_PATH $ENV{ARDUINO_SDK_PATH}/hardware/arduino/avr/programmers.txt)
> 
> set(PLATFORM_PATH         $ENV{PINOCCIO_PATH}/hardware)
> set(PLATFORM_ARCHITECTURE avr)
> 
> set(CMAKE_TOOLCHAIN_FILE $ENV{ARDUINO_CMAKE}/cmake/ArduinoToolchain.cmake)
> 
> [... MORE LINES ...]
> 

At that step i get:

>
> -- Generating blink
> CMake Error at /tmp/arduino/arduino-cmake/cmake/Platform/Core/BoardPropertiesReader.cmake:100 (message):
>   Board info not found: BoardName='pinoccio' BoardCPU=''
>   PropertyName='build.core'
> 

- Since the pinoccio hardware folder does not include any "core" folder (it uses arduino's default cores), i add the an extra variable for the core:

>
> set(PINOCCIO_CORES_PATH       $ENV{ARDUINO_SDK_PATH}/hardware/arduino/avr/cores)
>

but that does not help either...

aog2000a, 2018-01-06:
Summary:

- i needed my patch or modifying the boards.txt file to remove the "arduino:" piece.
- as soon as i define `PLATFORM_PATH` i needed to also define all 4 variables `ARDUINO_*_PATH`, even if the `ARDUINO_SDK_PATH` variable was still defined
- i needed to define `PINOCCIO_CORES_PATH` explicitly, i would expect that `ARDUINO_CORES_PATH` is used as fall back location

noseglasses, 2018-01-08:
> The boards.txt file allows the following syntax
pinoccio.build.core=arduino:arduino
to reference a core e.g. from the main Arduino SDK (see this for the detailed description). Currently arduino-cmake chokes on that syntax, since it preserves the full variable and can't find the corresponding folder.

That's odd as this problem was addressed by #19. The relevant change was kindly considered by MrPointer during his v2.0 refactoring. It ended up [here](https://github.com/arduino-cmake/arduino-cmake/blob/e377f9ca58313d34fcbf3a9cb0d550ff24f7b43e/cmake/Platform/Initialization/RegisterSpecificHardwarePlatform.cmake#L92). 

@aog2000a: Do you register the pinoccio platform via `cmake/Platform/Initialization/RegisterSpecificHardwarePlatform.cmake`? 
I am using a custom hardware as well that defines `model01.build.core=arduino:arduino` in `boards.txt and it works for me.

aog2000a, 2018-01-08:
@noseglasses  No, i just set the PLATFORM_PATH variable as mentioned in the README. My CMakeLists.txt has this piece:

> set(PLATFORM_PATH $ENV{PINOCCIO_PATH}/hardware/pinoccio)
> set(PLATFORM_ARCHITECTURE avr)
>
> set(CMAKE_TOOLCHAIN_FILE $ENV{ARDUINO_CMAKE}/cmake/ArduinoToolchain.cmake)

I actually debugged the issue by adding MESSAGE() lines in the arduino-cmake code, and could see that the "arduino:arduino" value in build.core was causing the 
`find_sources(CORE_SRCS ${BOARD_CORE_PATH} True)`
line in Platform/Core/Libraries/CoreLibraryFactory.cmake to fail.

noseglasses, 2018-01-08:
Please have a try with including `cmake/Platform/Initialization/RegisterSpecificHardwarePlatform.cmake` to register your custom hardware.

MrPointer, 2018-01-09:
@aog2000a You're right, currently there **is** a problem making it work for a platform like **pinoccio** that doesn't define any cores of it's but rather uses the Arduino's ones.
I'm currently working on a fix that will default to the Arduino hardware platform if required parts such as `core` and `bootloader` aren't found in the given custom platform.

MrPointer, 2018-01-11:
### Update
I've managed to add some code to resolve 3rd party platforms to the default **Arduino** platform automatically if required to, and it works perfectly when reloading CMake, however, fails on the build.
I don't have much time this weekend to address it further, maybe not even to share the build's failure reason. I hope that will change 😃

noseglasses, 2018-01-12:
> I've managed to add some code to resolve 3rd party platforms to the default Arduino platform automatically if required to

@MrPointer: My PR #55 contains a [commit](https://github.com/arduino-cmake/arduino-cmake/pull/55/commits/f441feaec7b1503eb558e985648007fb51435c81#diff-07edf1f8f0cf9043ccebd5dace3ac70cR45) that maybe related as it reads `platform.txt` (the very bottom of the commit in `RegisterSpecificHardwarePlatform.cmake`). If your approach is better, I will remove this commit.

MrPointer, 2018-01-14:
@noseglasses I took a look in that PR but it doesn't seem related to the issues I'm trying to solve.
Finally I got some time to finalize the code base so others can look at it and help further. Please take a look at my [Private Fork](https://github.com/MrPointer/arduino-cmake/tree/feature/3rd-party-platform-support) containing my work on the subject.

Basically I've added code that treats all 3rd party platforms as if they provide the entire 'package', but if not, like the **pinoccio** platform for example, it defaults to the native **Arduino SDK** in the missing parts. 
However, something's not right and I just can't figure out what - The CMake generation is fine yet the build is failing with the `fatal error: Arduino.h: No such file or directory` message, failing in the `IPAddress.cpp` file linkage. I've already encountered the error multiple times in the past, but never really understood what solved it...