arduino-cmake/arduino-cmake #52: Build project with ArduinoUnit.

Grandmother, 2018-01-07:
Hello, I'm trying to use arduino-cmake in clion. The problem appeared when I tried to compile my unit tests with arduino-cmake. I'm using [ArduinoUnit](https://github.com/mmurdoch/arduinounit) v2.0 in my 
tests. Even sketch from "getting started" doesn't compile.

First of all I renamed library folder from arduinounit to ArduinoUnit just to make arduino-cmake find it.
ArduinoUnit library was added to project like this:
```
link_directories(~/Arduino/libraries/)
```

The first I've got looks like this:
```
/usr/bin/cmake --build /home/grandma/projects/Arduino/testArduinoUnit/cmake-build-debug --target testArduinoUnit -- -j 2
[ 89%] Built target mega_CORE
Scanning dependencies of target testArduinoUnit
[ 92%] Building CXX object CMakeFiles/testArduinoUnit.dir/testArduinoUnit_testArduinoUnit.ino.cpp.obj
[ 96%] Linking CXX executable testArduinoUnit.elf
CMakeFiles/testArduinoUnit.dir/testArduinoUnit_testArduinoUnit.ino.cpp.obj: In function `_GLOBAL__sub_D_test_ok_instance':
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:12: undefined reference to `vtable for TestOnce'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:12: undefined reference to `vtable for TestOnce'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::~Test()'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `operator delete(void*, unsigned int)'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:12: undefined reference to `vtable for TestOnce'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:12: undefined reference to `vtable for TestOnce'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::~Test()'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `operator delete(void*, unsigned int)'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:12: undefined reference to `Test::run()'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::current'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::current'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::fail()'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::max_verbosity'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::min_verbosity'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::current'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::current'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::out'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::out'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::out'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::out'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::max_verbosity'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::min_verbosity'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::current'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::current'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::out'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::out'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::out'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::out'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `Test::out'
CMakeFiles/testArduinoUnit.dir/testArduinoUnit_testArduinoUnit.ino.cpp.obj:/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: more undefined references to `Test::out' follow
CMakeFiles/testArduinoUnit.dir/testArduinoUnit_testArduinoUnit.ino.cpp.obj: In function `_GLOBAL__sub_D_test_ok_instance':
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `TestOnce::TestOnce(__FlashStringHelper const*)'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:5: undefined reference to `TestOnce::TestOnce(__FlashStringHelper const*)'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:12: undefined reference to `Test::setup()'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:12: undefined reference to `TestOnce::loop()'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:12: undefined reference to `Test::setup()'
/home/grandma/projects/Arduino/testArduinoUnit/testArduinoUnit.ino:12: undefined reference to `TestOnce::loop()'
collect2: error: ld returned 1 exit status
make[3]: *** [CMakeFiles/testArduinoUnit.dir/build.make:100: testArduinoUnit.elf] Error 1
make[2]: *** [CMakeFiles/Makefile2:105: CMakeFiles/testArduinoUnit.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:117: CMakeFiles/testArduinoUnit.dir/rule] Error 2
make: *** [Makefile:131: testArduinoUnit] Error 2
```

Environment information:
```
[grandma@grandma_book ~/Arduino/libraries/ArduinoUnit/src ]$ pacman -Qi arduino
Name            : arduino
Version         : 1:1.8.5-1
[grandma@grandma_book ~/Arduino/libraries/ArduinoUnit/src ]$ pacman -Qi arduino-avr-core 
Name            : arduino-avr-core
Version         : 1.6.20-1
[grandma@grandma_book ~/Arduino/libraries/ArduinoUnit/src ]$ pacman -Qi avr-gcc 
Name            : avr-gcc
Version         : 7.2.0-1
[grandma@grandma_book ~/Arduino/libraries/ArduinoUnit/src ]$ pacman -Qi avr-libc 
Name            : avr-libc
Version         : 2.0.0-1

```

Grandmother, 2018-01-07:
Now I just tried to add ArduinoUnit to standard directory with libraries and added line
```
set(ArduinoUnit_RECURSE True)
```
to cmake/platform/Arduino.cmake.
Now I got the following output:
```
[ 43%] Built target mega_CORE
Scanning dependencies of target mega_Compare
Scanning dependencies of target mega_FakeStreamBuffer
[ 46%] Building CXX object CMakeFiles/mega_Compare.dir/usr/share/arduino/hardware/arduino/avr/libraries/ArduinoUnit/meta/mkCompare.cpp.obj
[ 46%] Building CXX object CMakeFiles/mega_FakeStreamBuffer.dir/usr/share/arduino/hardware/arduino/avr/libraries/ArduinoUnit/meta/mkCompare.cpp.obj
/usr/share/arduino/hardware/arduino/avr/libraries/ArduinoUnit/meta/mkCompare.cpp:1:10: fatal error: iostream: No such file or directory
 #include <iostream>
          ^~~~~~~~~~
/usr/share/arduino/hardware/arduino/avr/libraries/ArduinoUnit/meta/mkCompare.cpp:1:10: fatal error: iostream: No such file or directory
 #include <iostream>
          ^~~~~~~~~~
compilation terminated.
compilation terminated.
make[3]: *** [CMakeFiles/mega_Compare.dir/build.make:63: CMakeFiles/mega_Compare.dir/usr/share/arduino/hardware/arduino/avr/libraries/ArduinoUnit/meta/mkCompare.cpp.obj] Error 1
make[3]: *** [CMakeFiles/mega_FakeStreamBuffer.dir/build.make:63: CMakeFiles/mega_FakeStreamBuffer.dir/usr/share/arduino/hardware/arduino/avr/libraries/ArduinoUnit/meta/mkCompare.cpp.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:280: CMakeFiles/mega_Compare.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:354: CMakeFiles/mega_FakeStreamBuffer.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:149: CMakeFiles/testArduinoUnit.dir/rule] Error 2
make: *** [Makefile:144: testArduinoUnit] Error 2
```

MrPointer, 2018-01-07:
@Grandmother Thanks for contributing to **arduino-cmake**! We value it very much :)
Personally I never tried writing any unit tests for Arudino, and in fact I'm not an Arduino developer at all, just happened to take "ownership" of this great project and help maintain it.
I'll try to tackle your problem myself in the following days hoping to give you a quick solution. Something's telling me our bad documentation is yet again to blame... (We're in the process of addressing it)

MrPointer, 2018-01-14:
@Grandmother Sorry for keeping silent, I still couldn't find any time to check this. Will do in the following week though!

Grandmother, 2018-01-15:
I can put the simplest test to GitHub to save your time and boost this issue solving process)

MrPointer, 2018-01-15:
@Grandmother I'd like that very much, thanks!

Grandmother, 2018-01-24:
Hi. [Here](https://github.com/Grandmother/arduino_unit_cmake_test) is the simplest test case.
The ArduinoUnit library is included in source tree (I just got everything from src directory of project).
The project has two commits. In the initial commit we have no changes in ArduinoUnit tree.
In second commit I just moved ArduinoUnit.cpp to the root directory of library.

Try to build bot commits please.

MrPointer, 2018-01-26:
Hi, I'm currently focusing on re-writing the project's documentation so I'm really low on time.
I believe the documentation is one of the most important things in any project, and since the current documentation is all packed-up in a single ReadMe file and is also a bit outdated, the documenting process requires extra time & effort. 
In other words - I don't have time right now to address the issue properly (again), I hope that you understand. I'm doing my best to finish it as quickly as I can.

per1234, 2018-02-06:
The first problem was caused by the use of incorrect include syntax in the ArduinoUnit library. `#include "foo.h"` syntax should be used for internal includes of libraries. This is a common problem in Arduino libraries because if they are installed in a standard libraries folder the incorrect syntax doesn't cause a problem (as you later discovered).

The second problem is that code that was never intended to be compiled is being compiled. The files in the [`meta` subfolder of the ArduinoUnit library](https://github.com/mmurdoch/arduinounit/tree/master/meta) is used to generate library code, it is intended to be run on a computer, not a microcontroller.

This means that arduino-cmake has a different behavior than Arduino. This is explained here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#layout-of-folders-and-files

There are two different library formats supported by the Arduino IDE:

### 1.0 format
When a header file is present in the root folder of the library it is considered to be in 1.0 format.
>the source code is searched from the library root folder and the utility folder

### 1.5 format
No header file can be present in the root folder of the library. A library.properties file must be present in the root folder of the library for it to be recognized.
>The source code found in src folder and all its subfolders is compiled and linked in the user’s sketch.

So in neither case would the Arduino IDE ever attempt to compile the contents of the `meta` subfolder of the library.

noseglasses, 2018-02-06:
This is the CMake file that is responsible for finding libraries:

https://github.com/arduino-cmake/arduino-cmake/blob/develop/cmake/Platform/Core/LibraryFinder.cmake

You could add some debug statements `message("...")` to find out where/why Arduino-CMake picks up the meta folder. You could also add `arduino_debug_on()` to the top of your CMakeLists.txt file to get more verbose output during configuration.