arduino-cmake/arduino-cmake #59: Explicit use of -I<include dirs> prevents correct use of response files

noseglasses, 2018-01-24:
In many places in Arduino-CMake, include directories are explicitly added to target COMPILE_FLAGS properties using `-I...`. This is not the correct way it is supposed to be done with CMake. 

Include directories are either meant to be added with `include_directories(...)` (global) or `target_include_directories(...)` (target-wise).
When added correctly, include directory specifications can be added to response files that are passed to the compiler. This is especially important for project that link a high amount of Arduino libraries, thereby generating very long compiler command lines.

Most compiler command lines that are currently generated by Arduino-CMake are too long to be accepted by Windows, leading to strange errors in the line of "avr-g++.exe: error: CreateProcess: No such file or directory". This is the weird way of the system to state that a subprocess of the compiler cannot be spawned as the command line is too long.

A proper fix could be to collect include directories in various places instead of adding them directly as `COMPILE_FLAGS` and to add them to the generated targets using `target_include_directories(...)`, thus removing all explicit usage of `-I...`. This has to be done in various places for builds of firmare, libraries, examples, etc.

MrPointer, 2018-01-26:
Hmm, very interesting... May it be somehow related to the issues I had in #49 ? Look at my latest comment, I also tagged you there.

The fix you propose seems to be the right one, at least I myself don't see anything different. Could you fix it yourself?

noseglasses, 2018-01-26:
Not sure about the issues in #49. 

As explicit definitions of `-I...` are scattered all over Arduino-CMake, I do not dare to fix it at large scale as I cannot test the changes appropriately.

Although I cannot offer a fix for the overall problem, I can at least provide one that minimizes the number of `-I...` command line entries as a temporary fix. The idea is to take the final `COMPILE_FLAGS` property of each target, tokenize it with ' -' as separators, then remove any redundant entries from the list and reestablish `COMPILE_FLAGS`. As long as there are no strange compile flags that must occur multiple times, this works fine (I don't know of any such compiler flags for gcc). The compiler command line is thus drastically shortened. 

This is however not a fix for the response file issue as all `-I...` directives still remain in `COMPILE_FLAGS` instead of being added using `include_directories(...)`. But it is a great improvement as in my case it only enables me to build under Windows without exceeding the 32k limit on command line length.

I will submit a PR ASAP.

MrPointer, 2018-02-03:
Temporarily solved in #62, closing for now.

closed 2018-02-03 by MrPointer