arduino-cmake/arduino-cmake #67: A simple project, when compiled using arduino-cmake, does not fit on the board.

kigster, 2018-01-29:
I have been working on a project, which I extracted into a standalone repo.  It's based off the `arduino-cmake` repo and follows the same structure.

Inside is the `NeoCube.cpp` — a tiny piece of logic controlling NeoPixels using the `WS2812FX` library.  The library does mention that it's big, and even has a `#define` that says:

```c++
#define REDUCED_MODES 1 // sketch is too big for Arduino w/32k flash, so invoke reduced modes
```

Regardless, when this sketch is compiled using Arduino IDE, the image size ends up being about 80% of the maximum 32Kb. Here is the last few lines of Arduino IDE compile log showing the size of the image:

```bash
Compiling library "Adafruit_NeoPixel"
Using previously compiled file: /var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/libraries/Adafruit_NeoPixel/esp8266.c.o
Using previously compiled file: /var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp.o
Compiling core...
Using precompiled core
Linking everything together...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p  -o "/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/NeoCircle.ino.elf" "/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/sketch/NeoCircle.ino.cpp.o" "/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/sketch/WS2812FX.cpp.o" "/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/libraries/Adafruit_NeoPixel/esp8266.c.o" "/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/libraries/Adafruit_NeoPixel/Adafruit_NeoPixel.cpp.o" "/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/../arduino_cache_345435/core/core_arduino_avr_pro_cpu_16MHzatmega328_51f02b7210b938436b779d1c032618e1.a" "-L/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156" -lm
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0  "/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/NeoCircle.ino.elf" "/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/NeoCircle.ino.eep"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy" -O ihex -R .eeprom  "/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/NeoCircle.ino.elf" "/var/folders/tn/x23sslwx27q12kq5npk70qs80000gn/T/arduino_build_171156/NeoCircle.ino.hex"
Using library Adafruit_NeoPixel at version 1.1.5 in folder: /Users/kig/workspace/oss/arduino/sketches/libraries/Adafruit_NeoPixel 
Sketch uses 26524 bytes (86%) of program storage space. Maximum is 30720 bytes.
Global variables use 918 bytes (44%) of dynamic memory, leaving 1130 bytes for local variables. Maximum is 2048 bytes.
```

However, when compiled using `arduino-cmake`, the image suddently grows to 124% of the available RAM

```
 ↪   ~/workspace/oss/arduino/sketches/NeoCube-AC/build     cmake ..
-- ARDUINO Boards:
--                   menu:
--                    yun: Arduino Y
--                    uno: Arduino/Genuino Uno
--              diecimila: Arduino Duemilanove or Diecimila
--                   nano: Arduino Nano
--                   mega: Arduino/Genuino Mega or Mega 2560
--                megaADK: Arduino Mega ADK
--               leonardo: Arduino Leonardo
--            leonardoeth: Arduino Leonardo ETH
--                  micro: Arduino/Genuino Micro
--                esplora: Arduino Esplora
--                   mini: Arduino Mini
--               ethernet: Arduino Ethernet
--                    fio: Arduino Fio
--                     bt: Arduino BT
--             LilyPadUSB: LilyPad Arduino USB
--                lilypad: LilyPad Arduino
--                    pro: Arduino Pro or Pro Mini
--               atmegang: Arduino NG or older
--           robotControl: Arduino Robot Control
--             robotMotor: Arduino Robot Motor
--                  gemma: Arduino Gemma
--     circuitplay32u4cat: Adafruit Circuit Playground
--                yunmini: Arduino Y
--                chiwawa: Arduino Industrial 101
--                    one: Linino One
--                unowifi: Arduino Uno WiFi
--
-- ARDUINO Programmers:
--            avrisp: AVR ISP
--        avrispmkii: AVRISP mkII
--        usbtinyisp: USBtinyISP
--        arduinoisp: ArduinoISP
--     arduinoisporg: ArduinoISP.org
--            usbasp: USBasp
--          parallel: Parallel Programmer
--      arduinoasisp: Arduino as ISP
--          usbGemma: Arduino Gemma
--         buspirate: BusPirate as ISP
--            stk500: Atmel STK500 development board
[... 82 lines cut ...]
Scanning dependencies of target NeoCube
[ 97%] Building CXX object src/CMakeFiles/NeoCube.dir/NeoCube.cpp.obj
In file included from /Users/kig/workspace/oss/arduino/sketches/NeoCube-AC/src/NeoCube.cpp:3:0:
/Users/kig/workspace/oss/arduino/sketches/NeoCube-AC/src/libraries/WS2812FX/WS2812FX.h:424:30: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     uint8_t _segment_index = 0;
                              ^
/Users/kig/workspace/oss/arduino/sketches/NeoCube-AC/src/libraries/WS2812FX/WS2812FX.h:425:29: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     uint8_t _num_segments = 1;
                             ^
/Users/kig/workspace/oss/arduino/sketches/NeoCube-AC/src/libraries/WS2812FX/WS2812FX.h:429:5: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
     };
     ^
/Users/kig/workspace/oss/arduino/sketches/NeoCube-AC/src/libraries/WS2812FX/WS2812FX.h:429:5: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
[100%] Linking CXX executable NeoCube.elf
Generating EEP image
Generating HEX image
Calculating image size
Firmware Size:  [Program: 40788 bytes (124.5%) ]  [Data: 980 bytes (47.9%) ] on atmega328p
EEPROM Size:  [Program: 0 bytes (0.0%) ]  [Data: 0 bytes (0.0%) ] on atmega328p

```

This prevents me from using Arduino-CMake for this project, but I wonder what else does this affect?

Thanks so much!

noseglasses, 2018-01-29:
It seems that Arduino-CMake compiles stuff that is ignored by Arduino SDK. 

To find out what it is, you could use the following command to generate a list symbols for both builds, the Arduino SDK's and Arduino-CMakes's.
```bash
<your path to arduino>/hardware/tools/avr/bin/avr-nm -C <your firmware binary> | awk '{print $2" "$3}' | sort > symbols.txt
```

When you have two symbol files `symbols1.txt` and `symbols2.txt` you can simply diff the two files or use a tool like *meld* to compare them, to find out what is compiled where.

Once you know what it is that is compiled unnecessarily, you can try to eliminate it.

kigster, 2018-01-30:
Ok I’ll provide this info shortly

kigster, 2018-02-10:
I was able to run the command on both `elf` files, and then I removed the first column, and printed the rest sorted and uniqed. Here is the full command:

```bash
 avr-nm -C NeoCircle.ino.elf | awk '{printf "%s %s %s %s\n", $2, $3, $4, $5}'|sort | uniq > arduino.sorted.txt
```

Then I took a diff of the two files, first one being cmake version, and so the diff is below:

```diff
19d18
< B timer0_overflow_count  
30,36d28
< T Adafruit_NeoPixel::Adafruit_NeoPixel()  
< T Adafruit_NeoPixel::Adafruit_NeoPixel(unsigned int, unsigned
< T Adafruit_NeoPixel::Color(unsigned char, unsigned
< T Adafruit_NeoPixel::begin()  
< T Adafruit_NeoPixel::clear()  
< T Adafruit_NeoPixel::gamma8(unsigned char) const
< T Adafruit_NeoPixel::getBrightness() const 
38,41d29
< T Adafruit_NeoPixel::getPixels() const 
< T Adafruit_NeoPixel::numPixels() const 
< T Adafruit_NeoPixel::setBrightness(unsigned char) 
< T Adafruit_NeoPixel::setPin(unsigned char) 
43,48c31
< T Adafruit_NeoPixel::show()  
< T Adafruit_NeoPixel::sine8(unsigned char) const
< T Adafruit_NeoPixel::updateLength(unsigned int) 
< T Adafruit_NeoPixel::updateType(unsigned int) 
< T Adafruit_NeoPixel::~Adafruit_NeoPixel()  
< T HardwareSerial::_tx_udr_empty_irq()  
---
> T Adafruit_NeoPixel::show() [clone .constprop.39]
51,52d33
< T HardwareSerial::begin(unsigned long, unsigned
< T HardwareSerial::end()  
57,68c38,40
< T Print::print(Printable const&) 
< T Print::print(String const&) 
< T Print::print(__FlashStringHelper const*) 
< T Print::print(char const*) 
< T Print::print(char)  
< T Print::print(double, int) 
< T Print::print(int, int) 
< T Print::print(long, int) 
< T Print::print(unsigned char, int)
< T Print::print(unsigned int, int)
< T Print::print(unsigned long, int)
< T Print::printFloat(double, unsigned char)
---
> T Print::availableForWrite()  
> T Print::flush()  
> T Print::print(__FlashStringHelper const*) [clone
70,81c42,43
< T Print::println()  
< T Print::println(Printable const&) 
< T Print::println(String const&) 
< T Print::println(__FlashStringHelper const*) 
< T Print::println(char const*) 
< T Print::println(char)  
< T Print::println(double, int) 
< T Print::println(int, int) 
< T Print::println(long, int) 
< T Print::println(unsigned char, int)
< T Print::println(unsigned int, int)
< T Print::println(unsigned long, int)
---
> T Print::println() [clone .constprop.48]
> T Print::write(char const*) [clone
83,84d44
[... 269 lines cut ...]
> t WS2812FX::twinkle(unsigned long) 
> t WS2812FX::twinkle_fade(unsigned long) 
447,448d244
< t _GLOBAL__sub_I___vector_18  
< t _GLOBAL__sub_I_ws2812fx  
454d249
< t __empty  
460,461c255
< t _gammaTable  
< t _sineTable  
---
> t digital_pin_to_timer_PGM  
462a257
> t global constructors keyed
466a262
> t port_to_mode_PGM  
470,475d265
< t pstr_inf  
< t pstr_inity  
< t pstr_nan  
< t pwr_m10  
< t pwr_p10  
< t turnOffPWM  

```

kigster, 2018-02-10:
I put both elf files in the tar available here: [https://kig.re/downloads/NeoCircle-elf.tar.gz](https://kig.re/downloads/NeoCircle-elf.tar.gz).

fraillt, 2018-02-12:
Looks like there are no optimizations applied
maybe `cmake -DCMAKE_BUILD_TYPE=Release ..` would help?

kigster, 2018-07-04:
Unfortunately, adding these flags seems to make no difference:

```
$ rm -rf build && mkdir build
$ cd build && cmake -DCMAKE_BUILD_TYPE=Release ..
......
[100%] Linking CXX executable NeoCube.elf
Generating EEP image
Generating HEX image
Calculating image size
Firmware Size:  [Program: 40788 bytes (124.5%) ]  [Data: 980 bytes (47.9%) ] on atmega328p
EEPROM Size:  [Program: 0 bytes (0.0%) ]  [Data: 0 bytes (0.0%) ] on atmega328p

[100%] Built target NeoCube
```

kigster, 2018-07-04:
I've updated the [project](https://github.com/kigster/arduino-cmake-oversized-image) — please download it and run inside the project `./compile.sh` script.

lanouettea, 2018-07-06:
Hey @kigster ! I have been banging my head against the wall for a few hours trying to figure this one out on a project of mine in the past days. Turns out default compile flags do not strip unneeded symbols. I've checked the flags set by Arduino IDE and attempted to reproduce them on my end. Make sure you have the flags "-ffunction-sections -fdata-sections" added to your compiler flags. Here's what I have on my end:

`set(ARDUINO_CXX_FLAGS "-w -g -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics")
set(ARDUINO_LINKER_FLAGS "-w -g -fuse-linker-plugin -Wl,--gc-sections")`

MrPointer, 2018-07-07:
@kigster @lanouettea Hi all.

As written in the latest comments in #77, a new version of Arduino-CMake is coming.
In the new version, this exact problem of non-matching binary size when compiling with IDE vs compiling with Arduino-CMake is completely solved - **The compiler and linker flags are depending directly on the SDK instead of setting them manually**. It guarantees that with every change to the flags defined by the SDK, Arduino-CMake will update and behave accordingly.

But yet again - This version is yet to be published, and is still a _"Work In Progress"_, so you'll have to wait.

kigster, 2018-07-08:
@MrPointer You are my hero! It works!

Adding these flags makes the binary *almost* the same as when built with Arduino IDE:

**Arduino**:
  * Sketch uses **26524 bytes (86%)** of program storage space. Maximum is 30720 bytes  .
  * Global variables use 918 bytes (44%) of dynamic memory, leaving 1130 bytes for local variables. Maximum is 2048 bytes.

**CMake**
  * Firmware Size:  **[Program: 28288 bytes (86.3%) ]** [Data: 930 bytes (45.4%) ] on atmega328p
  * EEPROM Size:  [Program: 0 bytes (0.0%) ]  [Data: 0 bytes (0.0%) ] on atmega328p

Thank you *so fricken much* for this — honestly, it's been such a big bummer for me... You see, I wrote a small wrapper toolchain in Ruby, a tool called [arli](https://github.com/kigster/arli), which can generate a new project that uses Arduino-Cmake, can automatically download and install required libraries by the project based on `Arlifile.yml`, or it can be used as a CLI tool to search/install public Arduino libraries.

I even wrote my own [wrapper for Arduino-Cmake](https://github.com/kigster/arli-cmake/blob/master/cmake/Arli.cmake) to make building projects with lots of libraries easier. 

But all of that came to a halt when I discovered this incredibly mysterious issue — it has been a roadblock in moving Arli along, and building more projects with it. So I thank you because now I think I can just add these flags to `Arli.cmake` and it will do the right thing for all projects.

👍

kigster, 2018-07-08:
I had to change `-std=gnu++11` to `c++11`, and put it *before* I defined the toolchain. For quick reference, [here is the commit](https://github.com/kigster/arduino-cmake-oversized-image/commit/8306b544a92243455b0aba51d020fce253169252) to the repo that fixes the size issue.

kigster, 2018-07-09:
@MrPointer Is the new version on the develop branch? If so — are you interested in Beta testers?

lanouettea, 2018-07-10:
@kigster  I'm glad this helped you out! I too see a slight difference in file size, but this is negligible compared to the large discrepancies we had before. 
@MrPointer I'm looking forward to this new version. As kigster mentioned, if you need beta testers eventually, I'm in as well! Thanks for reviving this project and keeping it alive!

MrPointer, 2018-07-13:
@kigster @lanouettea 
Actually the new version is yet to be hosted on GitHub, and moreover - **It will reside under a new repository**. Why am I doing this? A few reasons:
1. There are way too many forks out there for this project, people are easily lost between them, so there's no need for another one.
2. In fact, the new project is not even a fork of an older one - It's a **complete fresh start**. 
This is because the older project (This one), as good as it got, is still a mess, and, more importantly, doesn't have the ability to address several critical issues that users complained about during the course of time due to its' core architecture.
3. It's time to move on and upgrade yourself. 
This projects guarantees to support Arduino **0.19!!!** This is insane! Nobody uses such old versions, and I'm really skeptical anyone uses a version older than 1.6, which came out in 2016.
Besides, the project also guarantees to support CMake 2.8, which is also very old and doesn't support many good features added by later version of CMake. So requiring a minimum of CMake 3.8, when even the apt package manager offers later versions, is totally fine.

As to your questions - Yes, I'd be happy to have you on-board and help me with some beta testings! 😄 
I will PM each of you when I'll have a functioning enough version.

MrPointer, 2018-09-11:
@lanouettea @kigster The new framework/repo is finally up & running!
Check it out [here](https://github.com/arduino-cmake/Arduino-CMake-NG) - Should solve this exact problem as well as many others.