arduino-cmake/arduino-cmake #73: Split files per platform

mean00, 2018-02-13:
Hi
I've made a proof of concept regarding adding very basic STM32 board support for arduino cmake
(other platforms would be "easy" to add after that).
Before going forward, there is something i'd like to propose that would make later pull request much easier:

split file per platform 

Some files are very tied to a specific platform : ArduinoToolchain.cmake, ArduinoBootloaderUploadTargetCreator etc etc

So the change would be to identify the platform (Avr, Stm32,...) and use that as a postfix
ArduinoToolchain.cmake just identifies the platform and includes ArduinoToolchain${platform}
Similarily, the other ones are now just include(foobar${platform}.cmake)

That way changes made to arduino-cmake will touch common files and **Avr.cmake file
and it will be much easier to track changes and merge them
Conversely, changes on other platforms would have zero impact on Avr  

Of course, any other changes with the same effects would be as good

Thanks

Ps: Proof of concept project https://github.com/mean00/arduino-cmake  , develop branch
Build my own Avr project + hello world on stm32

MrPointer, 2018-02-14:
@mean00 Thank you for investing time in this, and contributing again 😃 
While the idea is good, from my understanding it means that you plan to handle every platform on its own, and that is... not a good thing, even in a system like cmake which doesn't have real generics and such.
Take a look at #34 which discusses exactly that - We aim to develop it with "full-power" once the project finally gets stable enough. 
Basically, our current vision is to parse the `platform.txt` file of a custom platform, managing the build process and platform specific stuff based on the values there.

But maybe I'm wrong and your idea **is** better... @fraillt @noseglasses What do you guys think?

mean00, 2018-02-14:
Hi
The idea is to break the build into generic part and specific part
The generic part is common and shared by all platforms
The specific part is specific to each platform (compiler, linking uploading)

If you look at my Poc, you'll see that there was about 6 files changed and it now works fine on STM32 and  AVR now ( with the limited testing i've done), i've been playing with my STM32 for just a few days.

The main benefit is that it is  N simple problems
Someone fluent with ESP32 can do the porting, without breaking (not even touching) the AVR and STM32 parts. No risk of regression.

Pro : Simple
Cons : Not automatic, explicit work has to be added to support new platforms

The other approach is in principle better , but at the cost of complexity.
I.e. you have a single codepath  that needs to be able to manage everything

Pro : New platforms for free
Cons : Testing  + complexity

mean00, 2018-02-14:
Ah sorry, i thought you were mentioning the ESP32 support part.
Which, if i understood it correctly, was to better derive what to do from the arduino configuration files
(not sure i got it right)

noseglasses, 2018-02-15:
I would prefer to aknowledge Arduino's build files (platform.txt, etc.) as interface and concentrate on correctly parsing and supporting those. This way, by having as little platform specifics as possible mapped to CMake code, I expect Arduino-CMake to stay maximum extensible. Our long term aim would be to support any new platforms out of the box as long as their specification adheres to the Arduino standard.

mean00, 2018-02-17:
Fair enough. I'll close that issue, as it is now irrelevant.

closed 2018-02-17 by mean00