arduino-cmake/arduino-cmake #75: Board info not found during generation

DavidFair, 2018-02-20:
CMake fails to find a target for the `mega2560` target. I've tried with targeting the uno too which generates the same error message.
I've also tried specifying the BOARD and BOARDCPU together (which leads to the BOARDCPU argument not parsing and failing).

This works with the original project and compiles correctly, however I have not managed to get the fork work at all. The error is as follows:

```
CMake Error at ext/arduino-cmake/cmake/Platform/Core/BoardPropertiesReader.cmake:59 (message):
  Board info not found: BoardName='mega2560' BoardCPU=''
  PropertyName='build.core'
Call Stack (most recent call first):
  ext/arduino-cmake/cmake/Platform/Core/Libraries/CoreLibraryFactory.cmake:16 (_get_board_property)
  ext/arduino-cmake/cmake/Platform/Generation/ArduinoLibraryGenerator.cmake:28 (make_core_library)
  ext/CMakeLists.txt:13 (generate_arduino_library)
```

The CMakeLists.txt:

```
set(ENCODER_SRC
  encoder/Encoder.cpp)

set(ENCODER_HDR
  encoder/Encoder.h)

generate_arduino_library(encoder
    SRCS ${ENCODER_SRC}
    HDRS ${ENCODER_HDR}
    BOARD mega2560
)
```

- Ubuntu 16.04
- Arduino SDK version 1.0.5
- Latest commit on master

noseglasses, 2018-02-20:
Could you please specify what you mean by "the original project"?

It seems, you used the master branch of this repo. Could you please try the develop branch instead, which is a complete refactoring and the basis for future development.

MrPointer, 2018-02-21:
@noseglasses Well he isn't completely wrong is he? The `master` branch is set to the latest release, which should support `BOARD_CPU` seamlessly.
@DavidFair Personally, I suspect that the source of the problem is in your SDK version. Could you please update to the latest SDK and then try to rebuild the project?

noseglasses, 2018-02-21:
Good point. No, of course he is not wrong, when it comes to the master branch.

I am still curious about the original project because there are myriads of forks of the original arduino-cmake repo.

DavidFair, 2018-02-21:
I'm not sure what has changed but I did the following:

> It seems, you used the master branch of this repo. Could you please try the develop branch instead, which is a complete refactoring and the basis for future development.

After checking out `develop` it correctly generated the targets. However after checking out `master` to complete a bisect it now works.... so I'm not sure what has changed locally.

To answer the remaining questions:
> Could you please specify what you mean by "the original project"?

https://github.com/queezythegreat/arduino-cmake

> Personally, I suspect that the source of the problem is in your SDK version. Could you please update to the latest SDK and then try to rebuild the project?

I was about to check how SDK version affected CMake however now it works I cannot.

I'm going to close this issue since I can no longer replicate on either of the machines I am developing on. I suspect it was an issue with the git submodule.

DavidFair, 2018-03-05:
Reopening this:
` ... I suspect it was an issue with the git submodule. ` 
The git submodule kept pulling it back to commit a8b3761 which works hence the bisect did not work. Pointing to a normal git repo meant the bisect worked correctly.

This regression was introduced by: 
```
17d60adf04c92a2e24ad4da37ebc9ac63feaf506 is the first bad commit
commit 17d60adf04c92a2e24ad4da37ebc9ac63feaf506
Author: Timor Gruber <timor.gruber@gmail.com>
Date:   Fri Nov 24 15:28:56 2017 +0200

    Added ability to register custom hardware platforms.
    It's more a bug-fix combined with a feature, since the function to register a platform simply been inlined, defaulting 'Arduino' and 'avr' as platform and architecture.
    This commit allows developers to specify their own platform and architecture by setting the `PLATFORM_PATH` and `PLATFORM_ARCHITECTURE` variables BEFORE including the Arduino-CMake toolchain.
```

The previous commit correctly configures. After this commit the `ARDUINO_BOARDS_PATH` `ARDUINO_BOOTLOADERS` and `ARDUINO_CORES_PATH` are not found hence the previous failure I was getting.

As this commit is an ancestor to both master and develop it means both branches do not configure correctly for me.

This happens also on Ubuntu 17.10 as well as 16.04 in the original issue.