arduino-cmake/arduino-cmake #76: Joining forces on Arduino-CMake

noseglasses, 2018-02-21:
Github's repository fork diagram of Arduino-CMake (https://github.com/arduino-cmake/arduino-cmake/network/members) shows that there are currently more than 170 forks of the original project listed on Github. Some of these seem to have been abandoned, others appear to be under active development.

While our version of the project is not the one with the greatest number of sub-forks, it seems to be the only one that was significantly refactored. There are PRs and issues of users that compare this project to other versions of Arduino-CMake. Some users still base major changes on our master branch or even on other fork's master branches. This can cause trouble and frustration for them and for us and it often is a pitty to see that nice and valuable contributions cannot be easily merged to our develop branch.

Most people use Google to search for open source software and it is not obvious to them which version of Arduino-CMake is the most advanced or the best supported one. Many users might even be unaware that there are multiple versions of the same project at all.

It might therefore be a good idea to ask queezythegreat to add a note to the top of his README that informs about his version being inactive and notifying users about other project versions that are under active development. It would at least be a start to inform users that there is a multitude of forks and leave the choice to them.

Beyond that, wouldn't it be a good idea to try and join forces and work on a single Arduino-CMake project with as many contributors as possible?

This could be possible by contacting the users or organizations of (all) other Arduino-CMake forks, inviting them to join forces, contribute and use our version.

There are some sub-forks with a significant number sub-sub-forks, that might be the most promising to contact and talk to first.

https://github.com/francoiscampbell/arduino-cmake
https://github.com/altexdim/arduino-cmake

Does that make sense?

MrPointer, 2018-02-26:
I'm really sorry for being so late to the party, been really busy lately and still am.

I completely agree with you, and in fact I thought about it myself but never had the chance to actually do that. What bothers me the most is the fact that our project is not yet stable, and we don't have at single version that could actually be shipped with confidence. Until then, I'm not sure we can be taken seriously, especially by the original author *queezythegreat*.
BTW, I've opened an issue in the original repo to point out to ours, just to let people know a somewhat stable and updated fork exists. It did add some starts and even contributors, but not enough of course.

IMO, we should make our best efforts to stabilize the current project and reach version **2.1**, and then we could absolutely deal with external merging and "joining forces".
What do you think?

noseglasses, 2018-02-27:
Sounds good.

closed 2019-03-20 by noseglasses