arduino-cmake/arduino-cmake #77: Board info not found

Koekelas, 2018-02-24:
OS: Linux
Arduino SDK: 1.8.5-2017.01.09

# How to reproduce
```
git clone https://github.com/arduino-cmake/arduino-cmake.git
mkdir arduino-cmake/build
cd arduino-cmake/build/
cmake ..
```

CMake output:
```
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/avr-gcc
-- Arduino SDK version 1.8.5-2017.01.09: /usr/share/arduino
-- Check for working C compiler: /usr/bin/avr-gcc
-- Check for working C compiler: /usr/bin/avr-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/avr-g++
-- Check for working CXX compiler: /usr/bin/avr-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- 
-- Generating wire_example
CMake Error at cmake/Platform/Core/BoardPropertiesReader.cmake:59 (message):
  Board info not found: BoardName='uno' BoardCPU='' PropertyName='build.core'
Call Stack (most recent call first):
  cmake/Platform/Core/Libraries/CoreLibraryFactory.cmake:16 (_get_board_property)
  cmake/Platform/Generation/ArduinoLibraryExampleGenerator.cmake:34 (make_core_library)
  example/CMakeLists.txt:11 (GENERATE_ARDUINO_LIBRARY_EXAMPLE)

-- Configuring incomplete, errors occurred!
See also "/home/nicolas/10 Sorteren/tmp/arduino-cmake/build/CMakeFiles/CMakeOutput.log".
```

# What I tried
Before yesterday, I had never seen an Arduino or heard of CMake. I have no idea what I'm doing.

Looking through CMakeCache.txt, I noticed:
```
ARDUINO_BOARDS_PATH:FILEPATH=ARDUINO_BOARDS_PATH-NOTFOUND
ARDUINO_BOOTLOADERS_PATH:FILEPATH=ARDUINO_BOOTLOADERS_PATH-NOTFOUND
ARDUINO_CORES_PATH:FILEPATH=ARDUINO_CORES_PATH-NOTFOUND
ARDUINO_LIBRARIES_PATH:FILEPATH=ARDUINO_LIBRARIES_PATH-NOTFOUND
ARDUINO_PLATFORM_HEADER_FILE_PATH:FILEPATH=ARDUINO_PLATFORM_HEADER_FILE_PATH-NOTFOUND
ARDUINO_PLATFORM_LIBRARIES_PATH:FILEPATH=ARDUINO_PLATFORM_LIBRARIES_PATH-NOTFOUND
ARDUINO_PROGRAMMERS_PATH:FILEPATH=ARDUINO_PROGRAMMERS_PATH-NOTFOUND
ARDUINO_VARIANTS_PATH:FILEPATH=ARDUINO_VARIANTS_PATH-NOTFOUND
```

Which led me to:
```
$ ls /usr/share/arduino/hardware/
archlinux-arduino  platform.keys.rewrite.txt  platform.txt
```

archlinux-arduino is similar to the default Arduino platform in structure and contents:
```
$ tree -L 1 /usr/share/arduino/hardware/archlinux-arduino/avr/
/usr/share/arduino/hardware/archlinux-arduino/avr/
├── boards.txt
├── bootloaders
├── cores
├── firmwares
├── libraries
├── platform.txt
├── programmers.txt
└── variants
```

Just to be clear, I only have the archlinux-arduino platform installed.

I tried symlinking archlinux-arduino to arduino and adding:
```
set(PLATFORM_PATH ${ARDUINO_SDK_PATH}/hardware/archlinux-arduino/)
```
to CMakeLists.txt before including the toolchain file. Both had no effect.

Thank you in advance.

DavidFair, 2018-03-05:
This looks like a duplicate of #75 - could you do the following to check that it is the same commit which has regressed for you (I've adapted your original example):

```
git checkout 188a831
rm -r arduino-cmake/build
mkdir arduino-cmake/build
cmake ..
```
This should correctly configure

Then try
```
git checkout 17d60ad
rm -r arduino-cmake/build
mkdir arduino-cmake/build
cmake ..
```

This should fail? If so we are both experiencing the same issue

Koekelas, 2018-03-06:
Checking out 188a831 and building blink_example outputs:
```
CMake Error at cmake/Platform/Core/BoardFlags/FlagsSetter.cmake:33 (message):
  Invalid Arduino board ID (uno), aborting.
Call Stack (most recent call first):
  cmake/Platform/Core/Targets/ArduinoFirmwareTargetCreator.cmake:25 (set_board_flags)
  cmake/Platform/Generation/ArduinoExampleGenerator.cmake:52 (create_arduino_firmware_target)
  example/CMakeLists.txt:11 (GENERATE_ARDUINO_EXAMPLE)
```
archlinux-arduino is still symlinked to arduino.

qwattash, 2018-06-27:
I am having the same issue. I have the same arduino SDK version running on archlinux.

It seems that the archlinux installation of the SDK is weird as you already noted.
From what I understand, in https://github.com/arduino-cmake/arduino-cmake/blob/e377f9ca58313d34fcbf3a9cb0d550ff24f7b43e/cmake/Platform/Initialization/RegisterHardwarePlatform.cmake#L1-L3:
but the platform path in archlinux should be SDK_PATH/hardware/archilinux-arduino

The fast workaround of symlinking */usr/share/arduino/hardware/arduino* to */usr/share/arduino/hardware/archlinux-arduino* is working for me.

However I believe that a better solution should be to set the CMake PLATFORM_PATH variable.
Doing so
`cmake -DPLATFORM_PATH=/usr/share/arduino/hardware/archlinux-arduino`
appears to be causing an error when checking for valid variables:

```
CMake Error at cmake/Platform/Core/VariableValidator.cmake:17 (message):
  ARDUINO_CORES_PATH not set: Invalid Arduino SDK path (/usr/share/arduino).
```

This seems to be happening because in https://github.com/arduino-cmake/arduino-cmake/blob/e377f9ca58313d34fcbf3a9cb0d550ff24f7b43e/cmake/Platform/Initialization/RegisterSpecificHardwarePlatform.cmake#L5-L23
the core path is stored in `${PLATFORM}_CORE_PATH`, which expands to *ARCHLINUX-ARDUINO_CORE_PATH*.
Validation fails when in https://github.com/arduino-cmake/arduino-cmake/blob/e377f9ca58313d34fcbf3a9cb0d550ff24f7b43e/cmake/Platform/Initialization/TestSetup.cmake#L1-L5
the variable ARDUINO_CORE_PATH is checked, in place of ARCHLINUX-ARDUINO_CORE_PATH.
So to make the PLATFORM_PATH solution working it seems that we should patch the TestSetup file to correctly handle different platforms, although I am not sure if I am correct about this.

lanouettea, 2018-07-05:
I too am trying to get another board configuration to work with my arduino-cmake project. The board is a Dragino Yun Uno ("unoyun" board name). The board manager json is available here http://www.dragino.com/downloads/downloads/YunShield/package_dragino_yun_test_index.json.
I've installed it using the board manager, and I am now trying to point my CMakeLists.txt file to this location by using 

`set(PLATFORM_PATH /home/andre/.arduino15/packages/Dragino/hardware/avr/0.1.0)`

However, I do not see the board listed when calling print_board_list(). If I am right, it seems like arduino-cmake/cmake/Platform/Initialization/RegisterSpecificHardwarePlatform.cmake line 5 actually overwrite the PLATFORM_PATH variable. 

@MrPointer Do you have a stable-ish dev version of your new version I could try in the meantime?

MrPointer, 2018-07-07:
@lanouettea Unfortunately no, I still haven't reached a stable version which could be published and tested.
About the problems with the current repo - I honestly have no idea what's going on there, I'm sorry ☹️