arduino-cmake/arduino-cmake #79: Support CLion

Green0Photon, 2018-03-02:
The reason why I found out about this project was because of the CLion Arduino plugin. The plugin is mostly dead because arduino-cmake was dead. Now that this has been resurrected as an official project, I think it would make sense for this organization to potentially take that up too.

I have no idea whether anyone in this project even has the required CLion experience, but I figured I would let you guys know. If anyone would do it, it would make sense for you guys to. I know it's a little far reaching for a "feature" request, but I figured I should just put it out there. After all, there's no "organization issues" tab.

FYI, I'm here mostly because the Arduino IDE is terrible, which is the same reason this project exists...

MrPointer, 2018-03-03:
Hello there and welcome aboard! Glad you discovered our fork and decided to give it a try 😄 

As for your request - Personally, I use CLion to develop in general and this project specifically, and I can assure you that there are multiple other guys here that use it as well - So you're in good hands 😉 
Moreover, I thought about it myself as I stumbled across this plugin too, realizing it's quite dead.
I have absolutely no experience with writing plugins for CLion, but I guess they require a little bit of Java knowledge, as the entire IntelliJ platform (which CLion is based on) is written in Java. I do have some experience in Java, though I'm not sure it's enough to write a plugin, but what's more important here is that I personally do not wish to develop in Java right now, or in the near future.
But that's only me - There are many people out there who are, and probably want to see this feature implemented as we do.

Nevertheless, this feature is **completely out of the scope of this project**, and must be addressed in a separate repository, unique for that purpose. One drawback I can think of when doing this is of course the extreme loose coupling between the two, which can lead to "update lag", but that's definitely something I can live with.

Thanks for pointing it out though, I hope this will trigger some other developers to take it seriously 😊

necauqua, 2018-04-15:
For me, having Serial Monitor plugin and this project is enough for CLion.
Can i ask a question here, is it possible to hide all those Arduino lib files from showing in CLion as source files?
They are located in `/usr/share/arduino/hardware/*`, and CLion also complains about them being out of project folder.

Except that, everything works fine and CLion is much better than Arduino IDE, thank you very much for your work.

rileyrg, 2019-03-05:
Is this project still alive or is Arduino-CMake-NG the place to go. Not that I can get that working either of course ;)  Honest Q: is it beating a dead horse at this stage and better to just use platformIO? When I get "unknown" function for upload_arduino_target which is in their (arduino-cmake-ng) BASE example I have to wonder if there's any point.

MrPointer, 2019-03-07:
@rileyrg This project is completely **_stale and dead_**, it's written in **bold** in the Readme...
[Arduino-CMake-NG](https://github.com/arduino-cmake/Arduino-CMake-NG) is definitely your way to go now, as it replaces almost everything done by this project with newer & better code.