arduino-cmake/arduino-cmake #83: Example fails to build at `cmake ..`

marthinwurer, 2018-06-18:
Hey all, I used to use the original version, and just now tried to use this fork as it seemed to be the most up to date one after a cursory search.

I'm having issues building with this version. This computer was able to build a basic example fine with the original repo, but now that I'm using this fork it fails while running `cmake`. Here's the shell output:

```
$ cmake ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/avr-gcc
-- Arduino SDK version 1.0.5-2013.05.15: /usr/share/arduino
-- Check for working C compiler: /usr/bin/avr-gcc
-- Check for working C compiler: /usr/bin/avr-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/avr-g++
-- Check for working CXX compiler: /usr/bin/avr-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- 
-- Generating wire_example
CMake Error at cmake/Platform/Core/BoardPropertiesReader.cmake:59 (message):
  Board info not found: BoardName='uno' BoardCPU='' PropertyName='build.core'
Call Stack (most recent call first):
  cmake/Platform/Core/Libraries/CoreLibraryFactory.cmake:16 (_get_board_property)
  cmake/Platform/Generation/ArduinoLibraryExampleGenerator.cmake:34 (make_core_library)
  example/CMakeLists.txt:11 (GENERATE_ARDUINO_LIBRARY_EXAMPLE)

-- Configuring incomplete, errors occurred!
```

Any advice? I tried updating all the dependencies, and I'm using the Arduino SDK from the Ubuntu 16.04 repositories.

comfix, 2018-07-10:
Same problem occurring on ArchLinux.
Used Arch-repo Arduino SDK 1.8.5

MrPointer, 2018-07-13:
@marthinwurer @comfix 
Hi guys, please see my comment in https://github.com/arduino-cmake/arduino-cmake/issues/77#issuecomment-401338724 as it refers to a new version of Arduino-CMake that hopefully will solve your problems!

MrPointer, 2018-09-11:
@marthinwurer @comfix Hello again.
The new, updated framework is finally "online" and be found here: [Arduino-CMake-NG](https://github.com/arduino-cmake/Arduino-CMake-NG).
Although not all features have been ported to the new framework, you can comfortably try it!